sebanc/brunch #1816: No Speaker Audio on Acer 713-2w Brunch: r110 20230310

garyaburns, 2023-03-21:
I just wanted to start out by saying thank you to everyone that has worked so hard to get Brunch where it is today.  It is truly amazing to see all the hard work and effort that many have put into building something that is great.  

I have been wanted to dual boot my Acer Chromebook 713-2W with Windows10/11 and ChromeOS for a few years now, and I am finally very close to have both up and running.  

Here is what I currently have installed:
Chromebook:  Acer Chromebook Spin 713-2W
BIOS: Mr Chromebox 4.19
Chrome Image:  Chromeos_15329.44.0_hatch_recovery_stable-channel_mp-v6.bin
Brunch: r110 20230310
Kernel:  chromebook-6.1

From reading other post: Running the "cras" command from the Chromeos shell comes back with nothing.

I have the volume slider showing within ChromeOS and it shows an Output of Speaker (internal).  Just so sound coming out of the speaker.  Any help would be greatly appreciated.  I am a newbie to ChromeOS so please be patient with me and I will get you whatever information that may be helpful.  Thank you once again.

garyaburns, 2023-03-23:
In case this helps:

chronos@localhost / $ grep "" /sys/class/dmi/id/* 2> /dev/null
/sys/class/dmi/id/bios_date:01/29/2023
/sys/class/dmi/id/bios_release:4.19
/sys/class/dmi/id/bios_vendor:coreboot
/sys/class/dmi/id/bios_version:MrChromebox-4.19.1
/sys/class/dmi/id/board_asset_tag:
/sys/class/dmi/id/board_name:Kled
/sys/class/dmi/id/board_vendor:Google
/sys/class/dmi/id/board_version:rev7
/sys/class/dmi/id/chassis_asset_tag:
/sys/class/dmi/id/chassis_type:9
/sys/class/dmi/id/chassis_vendor:Google
/sys/class/dmi/id/chassis_version:
/sys/class/dmi/id/ec_firmware_release:0.0
/sys/class/dmi/id/modalias:dmi:bvncoreboot:bvrMrChromebox-4.19.1:bd01/29/2023:br4.19:efr0.0:svnGoogle:pnKled:pvrrev7:rvnGoogle:rnKled:rvrrev7:cvnGoogle:ct9:cvr:
/sys/class/dmi/id/product_family:Google_Hatch
/sys/class/dmi/id/product_name:Brunch
/sys/class/dmi/id/product_sku:sku0
/sys/class/dmi/id/product_version:rev7
/sys/class/dmi/id/sys_vendor:Google
/sys/class/dmi/id/uevent:MODALIAS=dmi:bvncoreboot:bvrMrChromebox-4.19.1:bd01/29/2023:br4.19:efr0.0:svnGoogle:pnKled:pvrrev7:rvnGoogle:rnKled:rvrrev7:cvnGoogle:ct9:cvr:

MagedMowafy35, 2024-05-03:
I have the same problem in the internal speaker and mic in Huawei Matebook D15 i5 11th gen, @garyaburns Did you find a solution to the problem?

MagedMowafy35, 2024-05-03:
I had tried to make the steps in troubleshooting list and h changed.

MagedMowafy35, 2024-05-03:
I had tried to make the steps in troubleshooting list and nothing changed.