sebanc/brunch #1818: Brunch r110 bad performance

pguarache, 2023-04-03:
Hi! I have months using using Brunch r107 on my Surface GO, the 8 gigas RAM version 128 GB internal storage, and everything is smooth, everything works great. When the r110 was released I decided to give it a try, brunch r110 and ChromeOS 110, and a clean install. But since the beggining everything is lagging, games closes, Google Play apps lags or stop responding. On version r107 I was able to play games like Wild Rift with no problem at all, on version r110 most games don't work smooth or even open... or close after it open. I used kernel 5.15, and tried 6.1 to see if the problem was solved and no luck. I switched back to r107 and everything work as usual, smooth with great performance.

Cheers.

iArti, 2023-04-07:
I'm using Surface Go too and I can confirm this issue
On brunch r110 and ChromeOS 110 system performance is unacceptibly bad
I used **top** command to find reason and I saw crosvm app using 100% of CPU and more than 5 Gb of memory all the time
It seems to be related with Android subsystem, but there is nothing I could do about it

Lorde627, 2023-04-13:
or you can try FydeOS.
https://fydeos.io/download/you/#microsoft

pguarache, 2023-04-13:
> or you can try FydeOS. https://fydeos.io/download/you/#microsoft

Fydeos Sucks terrible on Surfaces. Even the Surface "versions" they have. They use the Android x86 kernel and it Sucks. And paying? No thanks. I tried for a couple of hours and deleite It.

wokawoka, 2023-04-16:
I am experiencing the same issue

iArti, 2023-04-16:
I did some digging and found out that Brunch and ChromeOS r107 use **ARC++** for running Android apps. However, in r110 and newer Android apps use **ARCVM**. It's known for high CPU and memory usage. The only advantage is Android 11, whereas ARC++ is stuck with Android 9.
When I did a clean install of r110, system performance was unacceptable because ARCVM used all available RAM and 100% of the CPU (Pentium 4425Y in my case). So I installed r107 and the performance was really good.
Yesterday I updated from r107 to r111. Initially, it was barely usable, but after an hour or so, everything went back to normal. Android apps' performance is still not as good as it was with ARC++ (it required less resources to run), but it's mostly fine.
BTW, I use rammus leona ChromeOS image

bybenjamin, 2023-04-30:
> I did some digging and found out that Brunch and ChromeOS r107 use **ARC++** for running Android apps. However, in r110 and newer Android apps use **ARCVM**. It's known for high CPU and memory usage. The only advantage is Android 11, whereas ARC++ is stuck with Android 9. When I did a clean install of r110, system performance was unacceptable because ARCVM used all available RAM and 100% of the CPU (Pentium 4425Y in my case). So I installed r107 and the performance was really good. Yesterday I updated from r107 to r111. Initially, it was barely usable, but after an hour or so, everything went back to normal. Android apps' performance is still not as good as it was with ARC++ (it required less resources to run), but it's mostly fine. I don't know why such a major change was not reflected in changelogs, but maybe it's related more to Google rather than Brunch itself BTW, I use rammus leona ChromeOS image

Which image is using ARC++ android 9? ARCvm consumes too many resources

KANGKKKKKKK, 2025-01-22:
Now I'm trying 131, but the 64g RAM experience is not good, I'm trying to go back to 107, But version 107 doesn't exist on the restored image site

closed 2025-12-28 by github-actions[bot]