sebanc/brunch #1822: Brunch Internet Access

EvoGier, 2023-04-08:
Chrome OS boots up as normal, having me connect to the internet and such. However, when it comes time to add an account to the device, it becomes stuck on the "Please Wait" screen with the blue spinning circle. I am on Chrome OS 110. Nothing is an issue with Chrome os 100. But, when I decide to update the two of them, I become presented with this screen. Loading into the guest section works, but the internet does not.[image]

j-liaman, 2023-04-08:
me to

minjinmas, 2023-07-01:
I had the same truble on R113,R110,R107 when login the system after install ChromeOS with brunch framwork。
The wireless card was connected the wifi, but the internet update and test interface always running and running ....
I tried to press Control+Alt+F2 ,login:chronos  then ping www.google.com.hk,show ping sucessfull.
In the guest mode,the chrome browser can not work by input www.***.com
At one time,I want to give up ChromeOS.but google is very attractived.
lucky  in the  github issue feedback.  i know there is no above problems on Recovery img R100 .

minjinmas, 2023-07-01:
I had the same truble on R113,R110,R107 when login the system after install ChromeOS with brunch framwork。
The wireless card was connected the wifi, but the internet update and test interface always running and running ....
I tried to press Control+Alt+F2 ,login:chronos  then ping www.google.com.hk,show ping sucessfull.
In the guest mode,the chrome browser can not visit website by input www.***
At one time,I want to give up ChromeOS.but google is very attractived.
lucky  in the  github issue feedback.  i know there is no above problems on Recovery img R100 .ime,I want to give up ChromeOS.but google is very attractived.
lucky  in the  github issue feedback.  i know there is no above problems on Recovery img R100 .

closed 2025-12-28 by github-actions[bot]