sebanc/brunch #1826: Support new kernel surface linux

nhatminhhp1999, 2023-04-16:
In the new kernel release, surface linux team has added many important support for surface pro 8+9. I hope you can add the kernel to your project

peter-marshall5, 2023-04-18:
I would also be interested in this, since Brunch does not seem to support the touchscreen or the keyboard on my Surface Pro 9. (Brunch unstable with the option `ithc_touchscreen` does not work either.)

peter-marshall5, 2023-04-21:
I created [a fork of Brunch](https://github.com/peter-marshall5/brunch-unstable) that has fully working type cover and touchscreen support on my Surface Pro 9 in the latest release. I'm not opening a pull request because I deleted / slightly modified some kernel patches to get it to build (which might break Brunch on other devices), as well as the code being kind of sloppy.

sebanc, 2023-04-24:
Thanks a lot a @peter-marshall5, even though I applied the changes a bit differently your fork was very helpful.

Could someone try this brunch unstable build with kernel 6.1 ?
https://github.com/sebanc/brunch-unstable/releases/tag/r111-unstable-20230424

peter-marshall5, 2023-04-24:
I just tried it out, and everything is working properly. The battery power usage does seem to be a bit higher than on Linux, though. Strangely, the CPU clock jumps up higher than 3GHz and the power usage spikes insanely high when touching the screen. The same issue occurs on my own fork as well.

peter-marshall5, 2023-04-25:
Strangely, the issue with the CPU boosting to its max clock is no longer happening after disabling `81-power_settings.sh` in the Brunch patches folder

sebanc, 2023-04-26:
This is very interesting, however I don't really understand how 81-power_settings.sh patch could impact wakeups caused by ttyS0.

Did you maybe also remove 99-services.sh patch ? this one would be more likely to cause issues with ttyS0 and I am not actually sure if it is necessary.

peter-marshall5, 2023-04-26:
I tried messing around with `99-services.patch`, and I can confirm that disabling it causes the ttyS0 issue to appear. Just to clarify, the issue would happen when the system was idle. The touchscreen power usage is a separate issue.

peter-marshall5, 2023-04-26:
This is getting even stranger. Chrome OS seems to be accessing the touchscreen without iptsd. Touch still works after running `sudo stop ipts`, and `ps  -A | grep ipts` doesn't yield anything.

peter-marshall5, 2023-04-30:
I fixed the CPU frequency issue by setting the Intel HWP mode to `power` or `balance_power`. (It's set to `balance_performance` by default.)

This can be done by running `echo balance_power | sudo tee /sys/bus/cpu/devices/cpu*/cpufreq/energy_performance_preference
power`.

chivoyage, 2025-07-07:
> I fixed the CPU frequency issue by setting the Intel HWP mode to `power` or `balance_power`. (It's set to `balance_performance` by default.)
> 
> This can be done by running `echo balance_power | sudo tee /sys/bus/cpu/devices/cpu*/cpufreq/energy_performance_preference power`.?
The fork is dead, is there any way to still get this version for sp9 please?

closed 2025-12-28 by github-actions[bot]