sebanc/brunch #1829: How to set refresh rate on internal display?

peter-marshall5, 2023-04-20:
I'm running Brunch on a Surface Pro 9. `chrome://gpu` shows that the framerate is at 60Hz, and there seems to be some weird display artifacts as if the framerate is mismatched, similarly to how 60hz looks in Hyprland on Linux. The kernel cmdline option `video=2880x1920@120` seemingly has no effect. How can I set the framerate to 120hz?

peter-marshall5, 2023-04-21:
I got 120Hz to work with a custom EDID on Brunch unstable. (I set the pixel clock to zero in the section that specifies the 60Hz refresh rate, making it invalid.)

It would be nice if there was a way that worked on Brunch stable and without requiring special software or a hex editor.

thelife2face, 2023-11-02:
> I have the same problem with the Surface Pro 8 also just running at 60 fps

Hi bro, I can't Unable Touchscreen in Surface pro 8, Can you help me?
https://github.com/sebanc/brunch/issues/1958

wheeliemonster, 2023-11-22:
> I got 120Hz to work with a custom EDID on Brunch unstable. (I set the pixel clock to zero in the section that specifies the 60Hz refresh rate, making it invalid.)
> 
> It would be nice if there was a way that worked on Brunch stable and without requiring special software or a hex editor.

Hi. Would you be able to share how you created the custom EDID, please? A link or something will be fine.
I'm trying to lower the refresh rate on my Surface Pro 4 as this seems to somewhat mitigate the famous scramble/flicker issue.
Thanks.

peter-marshall5, 2023-12-12:
Sorry for the late response. I used a hex editor to manually edit a dump of the EDID, which uses the format described [here](https://en.wikipedia.org/wiki/Extended_Display_Identification_Data#EDID_1.4_data_format). [This site](http://www.edidreader.com/) can parse hex dumps of your EDID and help to validate manual changes made in a hex editor. There are also some proprietary Windows-only graphical EDID editors, but I didn't personally mess with any of them.

ThatStella7922, 2024-03-14:
Hi there, came across this for my SP9 and wanted to try as well, but I couldn’t seem to properly do the EDID edit (in VSCode’s hex editor) and im not sure how to apply the modified EDID either 

mind providing some pointers?

closed 2025-12-25 by github-actions[bot]