sebanc/brunch #1831: Brunch uses debug kernels in production

peter-marshall5, 2023-04-22:
I just wanted to point out that the Brunch build system configures kernels with `CONFIG_EXPERT` and various low-level debugging features enabled. Many of these seem to have no use whatsoever even in unstable releases of Brunch, and they could have an impact on performance and battery life.

sebanc, 2023-04-22:
CONFIG_EXPERT comes from the default chromiumos kernel config, android apps do not work without configs depending on this one.

peter-marshall5, 2023-04-22:
If you don't mind me asking, what are the configs required for Android apps
that depend on CONFIG_EXPERT?

sebanc, 2023-04-22:
I have not really tried to find out, I just know that when I disable CONFIG_EXPERT android apps stop working.

closed 2025-12-25 by github-actions[bot]