sebanc/brunch #1838: How secure is Chrome OS with Brunch?

evcode17, 2023-05-07:
Hello, I want to ask how secure is Chrome OS with Brunch? If I'm dual booting with Windows with Grub2Win and I don't enable secure boot (because Grub2Win doesn't support it), how secure is my Chrome OS installation and data?

tejasraman, 2023-06-04:
Secure boot != secure files - that only verifies the boot file to see if it matches a list of keys. Your data is pretty safe (ChromeOS encrypts days by default)

EvanEvan-17, 2023-06-19:
But does TPM on my PC works also on Chrome OS so my files will more secure?

Fede2782, 2025-01-22:
> But does TPM on my PC works also on Chrome OS so my files will more secure?

Brunch doesn't seem to use TPM. On the other hand, ChromeOS Flex DOES use TPM to enhance data protection