sebanc/brunch #1839: Crostini won't setup

Icey67, 2023-05-19:
Hello there,
I am currently running rammus 114 dev on brunch 111
And I can't install Linux ;-;

When I execute "vmc start termina" in crosh it says that a dlc service failed to install a termina dlc

lupohirp, 2023-05-27:
I have the same problem. Rammus 113 on brunch 111.

lupohirp, 2023-05-30:
I have an update regard to this problem. Doing a complete recovery image with  ChromeOS 113 base image, at first start i can install Crostini without any problem and start correctly. After a reboot i have this problem. 

I leave in attach the log of crostini VM 
[Crostini.log](https://github.com/sebanc/brunch/files/11605103/Crostini.log)

i found interesting this line: 
May 30 22:00:55 <11>  VM(33):  lxd[199]: t=2023-05-30T20:00:55+0000 lvl=eror msg="Failed to remove static IP network filters for \"eth0\": Failed to get a list of network filters to for \"eth0\": Failed to run: ebtables -L --Lmac2 --Lx: The kernel doesn't support the ebtables 'filter' table."#012

seems that is missing something in kernel

ivanzaqqa, 2023-06-10:
i have some problem on chromeos 113