sebanc/brunch #1843: wifi and bluetooth and trackpad not working on macbookair 2018

kawadaka, 2023-06-02:
wifi, bluetooth and trackpad not working on macbookair 2018!
How do I make it work?
For the time being, the version is r111!
I'm Japanese and I don't understand English, so I use google translate!
Thank you for your support!

tejasraman, 2023-06-04:
> wifi, bluetooth and trackpad not working on macbookair 2018!
> 
> How do I make it work?
> 
> For the time being, the version is r114 beta !
> 
> It didn't work with the r113 version either!
> 
> I'm Japanese and I don't understand English, so I use google translate!

Have you tried the MacBook kernel? The Wi-Fi should at least work there

MacBook Airs 2018 and later use a Force Touch trackpad, which might not have support... I'm not too sure though, will check later

(Japanese):

MacBookのカーネルを試したことがありますか？Wi-Fiは少なくともそこで機能するはずです

MacBook Airs 2018以降は、サポートされていないかもしれないフォースタッチトラックパッドを使用しています...よくわかりませんが、後で確認します

kawadaka, 2023-06-07:
I tried with macbook t2 kernel and driver wifi, bluetooth and touchpad don't work!

kawadaka, 2023-06-07:
If wifi, Bluetooth and trackpad do not work on chromebook, it will be fatal, so please improve it in the next version!

ZProLegend007, 2023-09-12:
Check this issue https://github.com/sebanc/brunch/issues/1914

:)

kawadaka, 2023-10-28:
Thank you for improving the driver problem after booting MacBook Air 2018 + Mac Mini 2018 with Chromeos. Thanks to you, wifi on macbookair 2018 and macmini 2018 and trackpad on macbookair 2018 are now working!

However, macbookair 2018 and macmini 2018
 I would like it to have Bluetooth, HDR support for internal and external monitors, and an earphone jack. YouTube is displayed in HDR, but when I check it on the monitor, HDR is turned off.
I know some HDR YouTube videos, so I'll post the links, so if you like, please check them out!

https://youtu.be/BeIeYibwwH4?si=En6miCLqtZBKHA8O

https://youtu.be/o0iMfOOei14?si=Qm9OkB8SEiqdwMyu

I can't say it's impossible, but...
Please support Touch ID operation and EGPU operation on MacBookAir 2018 and ChromeOS on MacBookAir M1 chip!

kawadaka, 2023-10-28:
Thank you for improving the driver problem after booting MacBook Air 2018 + Mac Mini 2018 with Chromeos. Thanks to you, wifi on macbookair 2018 and macmini 2018 and trackpad on macbookair 2018 are now working!
However, macbookair 2018 and macmini 2018
 I would like it to have Bluetooth, HDR support for internal and external monitors, and an earphone jack. YouTube is displayed in HDR, but when I check it on the monitor, HDR is turned off
I can't say it's impossible, but...
Please support Touch ID operation and EGPU operation on MacBookAir 2018 and ChromeOS on MacBookAir M1 chip!

kawadaka, 2023-10-28:
Thank you for improving the driver problem after booting MacBook Air 2018 + Mac Mini 2018 with Chromeos. Thanks to you, wifi on macbookair 2018 and macmini 2018 and trackpad on macbookair 2018 are now working!
However, macbookair 2018 and macmini 2018
 I would like it to have Bluetooth, HDR support for internal and external monitors, and an earphone jack. YouTube is displayed in HDR, but when I check it on the monitor, HDR is turned off.
I can't say it's impossible, but...
Please let me know how Touch ID works on macbookair 2018 and macmini 2018 and how egpu works!
m1 macbookair support using asahi linux please!

kawadaka, 2023-10-28:
Thank you for improving the driver problem after booting MacBook Air 2018 + Mac Mini 2018 with Chromeos. Thanks to you, wifi on macbookair 2018 and macmini 2018 and trackpad on macbookair 2018 are now working!
However, macbookair 2018 and macmini 2018
 I would like it to have Bluetooth, HDR support for internal and external monitors, and an earphone jack. YouTube is displayed in HDR, but when I check it on the monitor, HDR is turned off.
I know some HDR YouTube videos, so I'll post the links, so if you like, please check them out!
I can't say it's impossible, but...
Please make Touch ID and egpu work for macbookair 2018 and macmini 2018!
m1 macbookair support using asahi linux please!

ZProLegend007, 2023-10-28:
Hey kawadaka, I am currently looking for a fix for Bluetooth. I don't know much about how to fix the issue with HDR, it may be a chrome://flags setting that needs to be changed. Otherwise there may not be support for it. 

As for touch ID, it's almost impossible. Not even windows has support for the Mac touch ID. Same with M1 support, it's still a very large work in progress for linux. Integrating that with brunch is a very long way off.

As for eGPU, it should work with some configs that I will add to my repo and see if that helps. Otherwise you may need to do try some fixes yourself.

Here is my repo here: [Brunch-mac](https://github.com/ZProLegend007/brunch-mac)

kawadaka, 2023-10-29:
thank you.
Is it possible to start chromeos with secure boot on t2 mac using this method?

ZProLegend007, 2023-10-29:
Yes it is, follow the same steps on the original brunch repo for secure boot.

tejasraman, 2023-10-29:
> Yes it is, follow the same steps shown on the original brunch repo for secure boot.

The Z2 secure boot can be turned off in macOS recovery - it won't affect macOS. However, the method in the docs won't work for T2 Macs

ZProLegend007, 2023-10-30:
Ah I see, I was thinking there is not BIOS-like secure boot.

tejasraman, 2023-10-30:
> Ah I see, I was thinking there is not BIOS-like secure boot.

No, you need an actual installation of macOS to change the secure boot mode

ZProLegend007, 2023-10-30:
Or recovery mode.

ZProLegend007, 2023-10-30:
Question is, does it require a secure boot certificate?

tejasraman, 2023-10-30:
> Or recovery mode.

Well, for macOS, you need to actually have macOS installed to access the recovery partition (it's software, there is no way to change this in firmware settings (as there are none))

tejasraman, 2023-10-30:
> Question is, does it require a secure boot certificate?

No. It doesn't rely on regular certificates

kawadaka, 2023-10-30:
I would like to install ChromeOS with MacBook's secure boot enabled without disabling it! I know it's unreasonable, but I appreciate your help!

tejasraman, 2023-10-30:
> I would like to install ChromeOS with MacBook's secure boot enabled without disabling it! I know it's unreasonable, but I appreciate your help!

Secure boot isn't really more secure - and without a firmware modification (which is really hard with T2) it's not possible

kawadaka, 2023-10-30:
I'm sorry! 
Please improve so that HDR works!

tejasraman, 2023-10-30:
> I'm sorry! Please improve so that HDR works!

what does hdr have to do with T2 secure boot? And if HDR is broken, I can't help much

kawadaka, 2023-10-30:
It doesn't matter
I'm sorry!

closed 2025-12-25 by github-actions[bot]