sebanc/brunch #1844: Android subsystem and apps won't open

bybenjamin, 2023-06-02:
Everything works in fydeos, but not in android brunch and chromeos. No matter what I did. Please help me... @sebanc 
The structure I use is coral_111 and brunch_111
[image]
[image]

tejasraman, 2023-06-04:
@bybenjamin I know this is quite a simple step, but have you tried `android_init_fix` in brunch config yet?

bybenjamin, 2023-06-05:
I already tried what you said

lelemka0, 2023-07-29:
Have the same problem on coral_114 and brunch_114.
My device cannot use hardware acceleration when using rammus image, but coral works fine.
surface pro 3 with i3-4020y

tejasraman, 2023-07-29:
> Have the same problem on coral_114 and brunch_114.
> My device cannot use hardware acceleration when using rammus image, but coral works fine.
> surface pro 3 with i3-4020y

That's a different issue - it's related to some changes related to the removal of i915 in rammus. Just use coral, you'll be fine

lelemka0, 2023-07-29:
> > Have the same problem on coral_114 and brunch_114.
> > My device cannot use hardware acceleration when using rammus image, but coral works fine.
> > surface pro 3 with i3-4020y
> 
> That's a different issue - it's related to some changes related to the removal of i915 in rammus. Just use coral, you'll be fine

But the android subsystem doesn't work, I can't run any android apps after i switch to coral.

closed 2025-12-25 by github-actions[bot]