sebanc/brunch #1851: Brunch 113 unable to install Linux

Zece88, 2023-06-07:
Installation stops halfway an says an error has occurred.

Updates are "enabled"

Kernel version: 5.10

I where able to install Linux on older build, so something must be broken on the new build.

ArtemPozharov, 2023-06-09:
I had the same problem with Brunch v113. Now I am using Brunch v111 without any Crostini errors.
My config:
Google Chrome: Version 114.0.5735.119 (Official Build)  (64-bit)
Platform: 15437.42.0 (Official Build) stable-channel zork
Channel: stable-channel
Firmware Version: 
ARC Enabled: true
ARC: 10228649
Enterprise Enrolled: false
Developer Mode: true

Zece88, 2023-06-15:
> I had the same problem with Brunch v113. Now I am using Brunch v111 without any Crostini errors.
> My config:
> Google Chrome: Version 114.0.5735.119 (Official Build)  (64-bit)
> Platform: 15437.42.0 [...]

Could you write a little guide?

I assume that I should download Brunch v111 and "rammus recovery" matching v111 right?

Also, did you enable/disable something within flags/grub-config?

ArtemPozharov, 2023-06-15:
You can use the previous version of Brunch, but the most modern image of
Chrome OS. It is enough for a work.
You can contact me here: t.me/artem_pozharov

crestale, 2023-06-15:
Even upgrading to rammus 114 the problem remains.

ArtemPozharov, 2023-06-16:
Did you reinstall the system? Brunch v111 + Chrome OS v113?

crestale, 2023-06-16:
No. Brunch 113 and system 114.  I hope next brunch version.

Zece88, 2023-06-16:
[image]

Tried Brunch v111 with Chrome OS 113 (rammus).
Kernel: 5.15

Still no luck installing Crostini

ArtemPozharov, 2023-06-16:
Heb je virtualisatie gecontroleerd in het BIOS, mijn vriend uit Denemarken?

Zece88, 2023-06-16:
> Heb je virtualisatie gecontroleerd in het BIOS, mijn vriend uit Denemarken?
> 
> 
> > [image: PXL_20230616_105848717.MP.jpg]
> > <https://user-images.githubusercontent.com/64784816/246417250-7ccc8c86-5223-4f36-98b3-7d63056c0f61.jpg>
> >
> > Tried Brunch v111 with Chrome OS 113 (rammus).
> > Kernel: 5.15
> >
> > Still no luck installing Crostini

Yes[image]

ArtemPozharov, 2023-06-16:
Show me lscpu output, please. Ctrl+Alt+T -> shell -> lscpu.

Zece88, 2023-06-16:
> […](#)

Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
  Address sizes:         39 bits physical, 48 bits virtual
  Byte Order:            Little Endian
CPU(s):                  4
  On-line CPU(s) list:   0-3
Vendor ID:               GenuineIntel
  Model name:            Intel(R) Core(TM) i5-7500T CPU @ 2.70GHz
    CPU family:          6
    Model:               158
    Thread(s) per core:  1
    Core(s) per socket:  4
    Socket(s):           1
    Stepping:            9
    CPU(s) scaling MHz:  94%
    CPU max MHz:         3300.0000
    CPU min MHz:         800.0000
    BogoMIPS:            5399.81
    Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht 
                         tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cp
                         uid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4
                         _2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb inv
                         pcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 s
                         mep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln 
                         pts hwp hwp_notify hwp_act_window hwp_epp md_clear flush_l1d arch_capabilities
Virtualization features: 
  Virtualization:        VT-x
Caches (sum of all):     
  L1d:                   128 KiB (4 instances)
  L1i:                   128 KiB (4 instances)
  L2:                    1 MiB (4 instances)
  L3:                    6 MiB (1 instance)
Vulnerabilities:         
  Itlb multihit:         KVM: Mitigation: Split huge pages
  L1tf:                  Mitigation; PTE Inversion; VMX cache flushes, SMT disabled
  Mds:                   Mitigation; Clear CPU buffers; SMT disabled
  Meltdown:              Mitigation; PTI
  Mmio stale data:       Mitigation; Clear CPU buffers; SMT disabled
  Retbleed:              Mitigation; IBRS
  Spec store bypass:     Mitigation; Speculative Store Bypass disabled via prctl and seccomp
  Spectre v1:            Mitigation; usercopy/swapgs barriers and __user pointer sanitization
  Spectre v2:            Mitigation; IBRS, IBPB conditional, STIBP disabled, RSB filling, PBRSB-eIBRS Not affected
  Srbds:                 Mitigation; Microcode
  Tsx async abort:       Mitigation; TSX disabled

Zece88, 2023-06-16:
I can install Linux, but after I rebooted and disabled updates. Terminal wouldn't load[image]

Zece88, 2023-06-16:
Sadly all above has already been tried

ArtemPozharov, 2023-06-16:
Did you try to use 6.1 kernel?

crestale, 2023-06-17:
Oh yes

Zece88, 2023-06-17:
> Did you try to use 6.1 kernel?

Yes, no difference sadly

sebanc, 2023-06-17:
You could try resetting chrome sync:
https://support.google.com/chromebook/answer/2914794?hl=en

Make a backup of your synced data beforehand.

Zece88, 2023-06-18:
> You could try resetting chrome sync:
> https://support.google.com/chromebook/answer/2914794?hl=en
> 
> Make a backup of your synced data beforehand.
> 

Hi @sebanc ,

Thank you for replying on this issue.

I've tried what you recommended, but it made no difference.

I'm still unable to install and startup crostini

ArtemPozharov, 2023-06-18:
Does ARCVM work?

ArtemPozharov, 2023-06-20:
> ARCVM works. Seems that it has something to do with DLC, like is missing
> something in kernel. I tried also to change base img with ATLAS, but it
> doesnt' work. With ChromeOS Flex, i can install linux without any problem.

DLC is downloadable content?  How is this related to the kernel?

lupohirp, 2023-06-20:
> […](#)

should be. I don't know, but i think that is something related to kernel. The strange thing is that if you try to powerwash and then install linux,on first try it will works. But if you reboot, you will the problem again.

Could be useful if someone can post a dmesg, maybe the error is there....

Zece88, 2023-06-23:
> Why did you disable update?

Because if Chrome OS updates to a new build, before Brunch catch up, I will have a system which can't boot. I would much more like to control when it's safe to update without breaking something.

Also disabling updates should not be the reason for Crostini not being able to load after installation and a reboot, but I could be wrong

lupohirp, 2023-06-28:
anyone has updates on this problem? on 114 i has this problem yet

Zece88, 2023-06-29:
> anyone has updates on this problem? on 114 i has this problem yet

Yeah, this is still an issue

lupohirp, 2023-06-29:
> Also, did you enable/disable something within flags/grub-config

Didn't saw this message. I can confirm that the last woring release is ChromeOS 112. Starting from ChromeOS 113, Linux doesn't work anymore.

ArtemPozharov, 2023-06-29:
Try that:
https://www.reddit.com/r/Crostini/comments/kd1x0n/comment/go98xxw/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button

Zece88, 2023-06-30:
> Try that:
> https://www.reddit.com/r/Crostini/comments/kd1x0n/comment/go98xxw/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button

Still no luck

lupohirp, 2023-06-30:
Me too. I still have the problem. I have the same problem on other platform, an Intel nuc i5 11th gen. So, i think at this point that It Is a brunch problem

sebanc, 2023-07-01:
I have tried brunch on 3 devices using rammus, zork and volteer and could not reproduce this issue.

Is chromeos installed on an internal hdd or used on a usb hdd/flashdrive ?

Could you please attach /var/log/messages file after a failed crostini install ?

lupohirp, 2023-07-01:
> 

is installed in an internal hdd, an nvme ssd for accuracy.

here attached the log [messages.log](https://github.com/sebanc/brunch/files/11926765/messages.log)

Interesting thing could be this one?: 

`2023-07-01T17:19:55.444870Z ERR imageloader[5691]: Failed to open loop device: /dev/loop4096: No such file or directory (2)
2023-07-01T17:19:55.444883Z ERR imageloader[5691]: GetLoopDevice failed, mount_point: /run/imageloader/termina-dlc/package
2023-07-01T17:19:55.444889Z ERR imageloader[5691]: mount failed
2023-07-01T17:19:55.445209Z ERR dlcservice[1909]: ERROR dlcservice: [dlc.cc(701)] MountInternal(...): Domain=dlcservice, Code=failedToMountImage, Message=Imageloader LoadDlcImage() call failed.
2023-07-01T17:19:55.445234Z INFO dlcservice[1909]: INFO dlcservice: [dlc.cc(855)] Changing DLC=termina-dlc state to NOT_INSTALLED
2023-07-01T17:19:55.445374Z INFO dlcservice[1909]: INFO dlcservice: [metrics.cc(74)] InstallResult metric sent:9
2023-07-01T17:19:55.445383Z ERR dlcservice[1909]: ERROR dlcservice: [dlc.cc(605)] Mount failed during install finalization for DLC=termina-dlc
2023-07-01T17:19:55.445392Z ERR dlcservice[1909]: ERROR dlcservice: [dlc_manager.cc(191)] Install(...): Domain=dlcservice, Code=InternalError, Message=Failed to initialize installation for DLC=termina-dlc
2023-07-01T17:19:55.445400Z ERR dlcservice[1909]: ERROR dlcservice: [dlc_service.cc(128)] Failed to install DLC=termina-dlc
2023-07-01T17:19:55.445408Z ERR dlcservice[1909]: ERROR dlcservice: [dlc_service.cc(107)] [CoreServicesAlert<install>] Failed to install DLC=termina-dlc
2023-07-01T17:19:55.445441Z INFO dlcservice[1909]: INFO dlcservice: [metrics.cc(74)] InstallResult metric sent:9
2023-07-01T17:19:55.445451Z ERR dlcservice[1909]: ERROR dlcservice: [dlc_manager.cc(191)] Install(...): Domain=dbus, Code=org.chromium.DlcServiceInterface.INTERNAL, Message=Failed to initialize installation for DLC=termina-dlc
`

ArtemPozharov, 2023-07-01:
Sebunc, I have Chrome OS (zork) and Windows 11 installed in dual boot mode via Grub4Win. I have enabled automatic Chrome OS updates. Crostini can take a long time to load, and then issue "Launching vmshell failed: Error starting crostini for terminal: 51 (START_LXD_FAILED)."

sebanc, 2023-07-02:
Could you try to update brunch to this build ?
https://github.com/sebanc/brunch-unstable/releases/tag/r114-unstable-20230702

ArtemPozharov, 2023-07-02:
Grub2Win doesn't show the Brunch options in the list. I've followed all the
instructions.

lupohirp, 2023-07-02:
> Could you try to update brunch to this build ? https://github.com/sebanc/brunch-unstable/releases/tag/r114-unstable-20230702

with this unstable release, crostini is working now.

ArtemPozharov, 2023-07-02:
[image]

Zece88, 2023-07-02:
> https://www.reddit.com/r/Crostini/comments/kd1x0n/comment/go98xxw/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button

Confirms latest unstable brunch build solved my issue!
I was able to install Crostini, rebooted and it still worked.

@ArtemPozharov I left chrome://flags/#crostini-reset-lxd-db to "default"

In Grub I only have "PWA" + "Updates" enabled

Still need to test if disable updates makes it unusable again, but I guess not

Thank you all !!!

ArtemPozharov, 2023-07-02:
That's good. Do you use dual boot with Windows?

Zece88, 2023-07-02:
> That's good. Do you use dual boot with Windows?
> 
> 
> >
> > https://www.reddit.com/r/Crostini/comments/kd1x0n/comment/go98xxw/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button
> >
> > Confirms latest unstable brunch build solved my issue!
> > I was able to install Crostini, rebooted and it still worked.
> >
> > @ArtemPozharov <https://github.com/ArtemPozharov> I left
> > chrome://flags/#crostini-reset-lxd-db to "default"
> >
> > In Grub I only have "PWA" + "Updates" enabled
> >
> > Still need to test if disable updates makes it unusable again, but I guess
> > not
> >
> > Thank you all !!!

No dual boot. Only Brunch on internal nvme

Zedasix, 2023-07-13:
same problem, it's related to version 113

Zece88, 2023-07-14:
> same problem, it's related to version 113

Please try upgrade to this unstable release from sebach:

https://github.com/sebanc/brunch-unstable/releases/tag/r114-unstable-20230702

This worked for me and as the original post owner, I'm marking this issue as resolved

closed 2023-07-14 by Zece88