sebanc/brunch #1853: dGPU Problem

BilKomut, 2023-06-09:
Hi

I have HP 250 G6 laptop. 

i5 7200 CPU

AMD Radeon R5 M330

INTEL HD 620

I cant run brunch. I think this is because of dgpu. There is no option to disable it in the bios. What kernel command can I disable it with. Because the operating system boots up, comes the ChromeOS logo and then turns to black screen.

Thanks

ArtemPozharov, 2023-06-10:
You can try Ubuntu. In legacy mode, it installs without problems. When installing in UEFI mode, there will be a problem with the GRUB bootloader, but this is easily bypassed. Follow the instructions once and forget about the problems: https://bugs.launchpad.net/kernel/+bug/1835664/comments/31.
My mother has an HP 250 G6 laptop, so I know. But I installed Windows 10 LTSB 2016 on it. It works great.

SSP6904, 2023-09-18:
How is this related to the brunch bootloader exactly? He is talking about chromeOS, not Ubuntu or Windows 10.

neptalactone, 2023-10-25:
Hi, Idk if this will work for you, you can try to use custom experimental kernel command line refer from my issue post:
#1904