sebanc/brunch #1855: LG Gram 2023 issues - almost all fixed, but still No Sound. Can Someone Help?

mayaru, 2023-06-11:
Hi Everyone,

First of my gratitude goes to @sebanc for this wonderful project.
Now, talking about some issues that i have encountered with LG Gram 17Z90R, model 2023.
If you do own the laptop, it's a great ultrabook to use - I have had the 2021 model that worked with brunch everything out of the box - but with the newer 2023 model... well for 3 days now i have been banging my head to fix it. So the spare you some of the problems i faced, here it goes:

> Install latest brunch with brya recovery image following normal procedures documented by @sebanc.

> After the installation you will find some problems occuring:
1. minor (but annoying) screen flickering 
2. power issue: on closing the lid (sleep) will crash and restart
3. no sound in the internal speakers: no audio sound card is detected, the the volume bar is gray. 
*Audio through USB-C headphones, including mic, works perfectly though.

Atempting to fix this I found solutions for the first 2 problems and still trying to fix the sound problem:
Use Brunch configuration menu and:

1. select "i915.enable_fbc=0 i915.enable_psr=0" - will fix the screen flickering
2. select chromebook-6.1 Kernel - will fix the sleep problem
3. selecting snd-intel-dspcfg.dsp_driver=1 - will get the sound volume working, but still NO SOUND from internal speakers.

So... numer 3 is open for new suggestions.
About the laptop - it's a intel 13th gen processor, the the sound card is a realtek alc265 i think.

if someone has some ideas please let me know. thank you.

mayaru, 2023-06-12:
Solution numer 2:
https://forum.manjaro.org/t/howto-set-up-the-audio-card-in-samsung-galaxy-book/37090

But here i don't know how to install sof-firware on chrome os...
maybe someone or @sebanc can give a hand?

Thank you

nokidevz, 2023-06-17:
about the sound problem, i had the exact same issue, i fixed it by shutting down my laptop and turning it on, when i saw the brunch boot menu (or the Grub2Win Menu if you dualboot) you click the ,,e" buttuon on your keyboard before ChromeOS boots, like right in the dualboot menu or brunch menu, then you will see the grub config, in there just find something saying options=$options , there add the following so it looks like this: "options=$options pwa enable_updates chromebook_audio" without the " . after doing this make sure both areas where there is options=$options are filled the exact same. after doing this hold CTRL key and click on X . (CTRL+X) then boot into ChromeOS, then open ChromeOS Settings and find About ChromeOS, and click Check for updates, the update is about 1.4GB after updating check if the audio works. If it doesnt then try to update Brunch to an Unstable Version, and see. To do that you just download an unstable .tar.gz brunch, open terminal (CTRL+ALT+T) and write ,,shell" then write "sudo chromeos-update -f *.tar.gz" , all without " and replace the * with whatever the brunch unstable file is named as, then just reboot and repeat all of the steps, as when updating brunch i think the config resets and check again. Hope this helps!

mayaru, 2023-06-18:
Hi nokidevz,

Thank you for your detailed response.
Can you tell me which laptop you have? also LG Gram 17 2023 model?

nokidevz, 2023-06-18:
I have Asus X515EA (11th Generation) , After the update its ChromeOS 114

mayaru, 2023-07-18:
Just upgraded to Chroma OS R114 using the pwa-update. Overall no improvements noted, and on the downside, the suspend function does not work anymore. the fix was to use the chomebook kernel 6.1 and this was fixing it in R113, but now with R114 using chomebook kernel 6.1 will not have it work... also it does not recognizce the wifi card.
So, i will not recommend the upgrade for LG Gram 2023.

maybe @sebanc has some ideas on this. 🙏

ericmiranda7, 2023-09-21:
Please check out this comment solution (the script file is provided in a later comment)
[https://discussion.fedoraproject.org/t/lg-gram-2023-and-alsa-sof-firmware-sound-issues/87576/2](url)

Works great for me! Only you have to rebalance the audio slightly to the left, besides that it seems working in system sounds, spotify, browser, etc.

mayaru, 2023-10-03:
@ericmiranda7 thanks for the tip.
Could you provide a step by step instruction how you applied the fix? That would be greatly appriciated by everyone who is not so tech savy. Thank you.

closed 2025-12-25 by github-actions[bot]