sebanc/brunch #1862: installation error: your password could not be verified. Please try again. Could not mount cryptohome.

nkcong206, 2023-06-17:
[image]
my laptop is a Dell Vostro 3559, core i5 i5-6200U 6th skylake.
I get this error when trying to sign in with my Google account. I confirmed over the phone so it can't be the wrong password. I did a search and it is TPM related, I went into the BIOS menu and turned it on. I checked on Windows it was confirmed as TPM 2.0. However, the error is still not resolved. I can still log in generally with ChromeOs Flex. Help me, please!

antoinepod, 2024-12-18:
Hello, I have the exact same issue on Dell XPS 13 Plus 9320, and changing kernel version doesn't fix my problem 🙁 please help, I can't configure Chrome OS at all because of this

laconeo, 2025-02-24:
Hello I have a Lenovo Ideapad slim 3 14amn8 with a Rayzen 3, I have tried many things even to pass additional params in the kernel to deactivate the crypto home but nothing works

PeterDD9PS, 2025-05-15:
Try a different recovery file!  With the recovery file “sentry” I had the same error. For my Lenovo Yoga 260, the recovery file “nami” works perfectly without any errors!!!

I initially used the recovery file “sentry” for a Lenovo ThinkPad Yoga 260 (Intel® Core™ i7-6600U, Intel® HD Graphics 520), because its hardware is very similar to the Lenovo Thinkpad 13 Chomebook. But I could not fix the error “Could not mount cryptohome” with this recovery file. Only a guest login works. The Google account can then be accessed with the Chrome browser. If you then log out and try to set up an account again, the error “Cyptohome corrupted” appears. There is no way to fix the error with “sentry”.

My solution: 
I have tried several other recovery files. Of course, the recovery file “reven” works in combination with the brunch framework. The login also works here - even with the TPM 2.0 module deactivated. However, the PlayStore is missing in “reven”. You can also install the recovery file “reven” without the brunch framework. But with the framework, the root shell works via “CRTL-ALT-F2” and the developer mode also works.

For my Lenovo Yoga 260, the recovery file “nami” works perfectly!!! This recovery file also supports the hardware of my Yoga 260 and, unlike “sentry”, the error message did not appear, i.e. I was able to log in successfully:

[image]

closed 2025-12-25 by github-actions[bot]