sebanc/brunch #1863: sound not working

leatherknob, 2023-06-17:
i have an HP Elitebook x360 1040 G5 with Conexant HD audio, tried the chromebook audio patch and i still dont have audio output on speakers, everything else works fine (including audio over bluetooth)  

im using brunch r113 and recovery rammus 113

nokidevz, 2023-06-17:
i had the exact same issue, i fixed it by shutting down my laptop and turning it on, when i saw the brunch boot menu (or the Grub2Win Menu if you dualboot) you click the ,,e" buttuon on your keyboard before ChromeOS boots, like right in the dualboot menu or brunch menu, then you will see the grub config, in there just find something saying options=$options ,  there add the following so it looks like this: "options=$options pwa enable_updates chromebook_audio" without the " . after doing this make sure both areas where there is options=$options are filled the exact same. after doing this hold CTRL key and click on X . (CTRL+X) then boot into ChromeOS, then open ChromeOS Settings and find About ChromeOS, and click Check for updates, the update is about 1.4GB after updating check if the audio works. If it doesnt then try to update Brunch to an Unstable Version, and see. To do that you just download an unstable .tar.gz brunch, open terminal (CTRL+ALT+T) and write ,,shell" then write "sudo chromeos-update -f *.tar.gz" , all without " and replace the * with whatever the brunch unstable file is named as, then just reboot and repeat all of the steps, as when updating brunch i think the config resets and check again. Hope this helps!

leatherknob, 2023-06-18:
> i had the exact same issue, i fixed it by shutting down my laptop and turning it on, when i saw the brunch boot menu (or the Grub2Win Menu if you dualboot) you click the ,,e" buttuon on your [...]

will try it first thing in the morning 

thanks a lot !

nokidevz, 2023-06-18:
no problem, tell me if it works!

ly9999999, 2023-09-03:
After downloading the sof firmware, refer to this method. https://github.com/sebanc/brunch/issues/1859#issuecomment-1689698613

pvlpsh, 2024-02-14:
My chromebook is Samsung Chromebook Pro, I instaled branch 120, rammus 120. Everything works very well, except the sound. I tried different kernels (6.6, 6.1, 5.15, 5.10, 4), it doesn’t work. Sound patch for Chromebook Old Audio cannot find caroline.tar.gz. I dumped drivers from Chrome OS 114, copied them to the UCM for this inst - https://www.reddit.com/r/GalliumOS/comments/6ys3c0/implementing_this_audio_fix_on_chromebook_pro/ But it doesn’t work. It becomes possible to change the volume, but there is still no sound. Help me plz. On Linux Fedora 39, I started audio with  script after installing Linux according to the instructions, everything works. The drivers that I dumpd from 114 Chrome OS - sklnau8829adi. Help me plz. @sebanc