sebanc/brunch #1875: Battery indicator issues

nifsta, 2023-06-27:
Sorry in advanced as I'm new to this, took this on as a pet project, 

I've installed ChromeOS on my Dell Inspiron 15 7000 laptop (7th Gen i7), when the charger is plugged into the device, the indicator doesn't show that it's charging, even though it is (and the battery life just increases but it doesn't SHOW that it's charging)

Any ideas on how I can fix this?

gena01, 2023-09-14:
I have a reproducible case on 2 different laptops. This started happening several brunche/ChromeOS releases ago (it is working on 107 though).. Latest I tried  113 and 115 (latest stable brunch) release.

Let me know if you need more information or want me to try something?

tungmeister, 2023-11-19:
This has been an issue for a while but unfortunately, it's not getting fixed for some reason. 
https://github.com/sebanc/brunch/issues/1792
https://github.com/sebanc/brunch/issues/1770

mrme8895, 2023-12-16:
I think the problem is related to the chrome OS power manager 'powerd' but ,I don't know if it can be replaced or if there is a setting that is interfering with the hardware. I'm having the same issue and i'll update you as soon as i have found a soulution.

mrme8895, 2023-12-16:
> I have a reproducible case on 2 different laptops. This started happening several brunche/ChromeOS releases ago (it is working on 107 though).. Latest I tried 113 and 115 (latest stable brunch) release.
> 
> Let me know if you need more information or want me to try something?

Is this chromeos 107 or brunch 107 or both

tungmeister, 2023-12-16:
I first encountered the issue back in January.  I'm not certain when the issue arose as I'd not been using chomeos on this laptop for a while. Looking at the releases I'm guessing the problem started for me with R94 as earlier in 2022 my battery status was reported correctly.

mrme8895, 2023-12-17:
Try running `sudo initctl stop powerd` in crtl+alt+f1 logged in as root. Log out by using the `exit` command then sign out of your user session. Restart and then hit previously mentioned key combo. Enter `sudo initctl start powerd` and restart your system.

This worked for me.

tungmeister, 2023-12-17:
Sorry, I'm not following, surely after a restart powerd will already be running from startup?

mrme8895, 2023-12-17:
It wasn't running when I checked with `initctl list | grep powerd
` so I started it it again. If it isn't running start it back before logging in.

mrme8895, 2023-12-17:
#1770 I see here the battery test failed. Use these commands to add `chronos` to `cros_ec-access` group.
`sudo usermod -a -G cros_ec-access chronos` and run the battery test again. I'm not sure but it may also fix the battery indicator issues.

mrme8895, 2024-03-03:
Have you tried switching to kernel 5.15 or 5.10.  If these kernels have support for your hardware then you can switch to an older version by  going into the brunch settings. Do the following to switch kernels:
1. Hit the key combo ctrl+alt+f2.
2. Login as root 
3. Type `sudo edit-brunch-config`  
4. Select the 5.15 or 5.10 kernel 
5. Skip the rest by hitting enter or select any other options you may have previously had.

biplobsd, 2024-06-15:
I made a script for auto battery saver on/off when the power supply changed..
```bash
chronos@localhost /mnt/stateful_partition/brunch/bootscripts $ ls
auto_remount_nodiscard.sh  restart_prog.sh  zauto_battery_saver.sh
chronos@localhost /mnt/stateful_partition/brunch/bootscripts $ sudo cat zauto_battery_saver.sh 
#!/bin/bash

# Monitor power supply changes
dbus-monitor --system "type='signal',interface='org.chromium.PowerManager',member='PowerSupplyPoll'" |
while read -r line; do
    state=$(run-on-ac false && echo "1" || echo "0")

    # Set battery saver mode based on power source
    dbus-send --system --dest=org.chromium.PowerManager --type=method_call --print-reply /org/chromium/PowerManager org.chromium.PowerManager.SetBatterySaverModeState array:byte:8,$state
done
```