sebanc/brunch #1876: Tablet mode oddities

dch82, 2023-06-28:
Hello,

For some reason tablet mode is strange.
Right now for me it is just the normal desktop mode but with n added desks button.
The expected should be the default tablet mode.
Please check out the screenshot, especially the bottom right corner.

[image]

Thanks!

darkza, 2023-08-14:
same here if i press , ctrl  alt shift t, the multidesktop adds but not the tablet mode.
But if i remove the type cover Tablet mode works fine. 

Would be cool if i can change Tablet mode on key Press like the Accesibility menu!

On Flex ive got the Strange Bug that if i Start without type cover it wont activate Tablet mode....
Great.. how to Press Keycombo without Keyboard...

dch82, 2023-08-14:
@darkza Me too... I'm using a Surface Pro 2 instead though.