sebanc/brunch #1879: HP Elite C1030 No Sound

ffyter, 2023-07-04:
I have an hp Chromebook. I installed Brunch with the Jinlon 113 release. Everything works except for audio on the internal speakers. Tried all the kernals and have selected native Chromebook in the options. Verbose mode doesn't show any errors that I could see. Tried the unstable release of brunch and it wouldnt boot. Any help would be appreciated.

ffyter, 2023-07-04:
I actually do see no soundcard in verbose mode.

ly9999999, 2023-09-03:
After downloading the sof firmware, refer to this method. https://github.com/sebanc/brunch/issues/1859#issuecomment-1689698613

DexterSLamb, 2023-09-07:
I tried r115 build with chromebook 5.15 kernel and native_chromebook_image option. The headphone jack with earphone works great but no sound on speaker.
Any advise?
Thanks.

chenxijun, 2023-10-23:
I'm excited to share with you about my solution upon this issue.

HP Elite C1030 use `sof-rt5682` sound card, and I found its ucm lying in `/usr/share/alsa/ucm/sof-rt5682.base` which means the card should work. However, with `sudo alsactl info` showing me that the card's "CardName" is `sof-rt5682` but not `sof-rt5682.base`, alsa couldn't load the ucm properly. (PS: For why you can look for the explanation in `/usr/share/alsa/ucm2/ucm.conf`, which shows the searching path of ucms)

So here comes the solution, just follow these commands in your tty:
```
sudo mkdir /usr/share/alsa/ucm/sof-rt5682
sudo cp /usr/share/alsa/ucm/sof-rt5682.base/sof-rt5682.base.conf /usr/share/alsa/ucm/sof-rt5682/sof-rt5682.conf
sudo cp /usr/share/alsa/ucm/sof-rt5682.base/HiFi.conf /usr/share/alsa/ucm/sof-rt5682/HiFi.conf
```
then restart your chromebook and it works.

This might be helpful for other chromebooks that use similar soundcard.
For reference, I found ucm2, which is newer, from here: <https://github.com/alsa-project/alsa-ucm-conf/blob/master/ucm2/sof-soundwire/rt5682.conf>

DexterSLamb, 2023-10-23:
> I'm excited to share with you about my solution upon this issue.
>
> HP Elite C1030 use `sof-rt5682` sound card, and I found its ucm lying in `/usr/share/alsa/ucm/sof-rt5682.base` which means the card [...]

Great it works! I have been looking for the solution to solve this issue for months then finally I got here :)
谢谢

ffyter, 2024-03-10:
I can confirm that the above fix works to fix sound on my hp c1030.  @sebanc  any chance we could get this added to the next release?

closed 2025-12-23 by github-actions[bot]