sebanc/brunch #1889: Brunch 114 and ChromeOS 114 Playstore will not start

Dnalsi, 2023-07-20:
Used Brunch since 110 with everything working.  Using Rammus Shyvana.  Upgraded to both brunch and chromeOS to 114 and playstore just will not start, which also means I cannot get in to clear cache with android preferences.  Everything else works.  Intel based on 7th gen. Running secure boot. Things I have tried. 
Default config.
Android fix 1
Android fix 2
Removing cros_debug
Powerwash
Using Nami Syndra which used to work but does not at 114
Roll back ChromeOS to rammus shyvana 113 and left Brunch at 114 and all works again.

Any ideas?  I am stumped.
Thanks

Dnalsi, 2023-07-20:
I guess I should mention I allowed 114 to sit overnight with updates turned on and it took the 114 point release from 114....57 to ...114...63.  Store still just sat spinning at starting Playstore.  This is triple boot Win11 and Linux 21.2, not that that should take a difference.

ArtemPozharov, 2023-07-24:
I have the same problem

lupohirp, 2023-07-27:
Chrome OS 115, ARCVM doesn't work anymore. On 114 works flawlessy

ArtemPozharov, 2023-07-27:
I confirm your words.

lupohirp, 2023-07-27:
any way to check arcvm logs?

sebanc, 2023-07-31:
I can reproduce the issue but I don't understand where it comes from, for now the workaround is to disable android and re-enable it.

fabiosscardoso, 2023-09-04:
Hi. Using my dell G3 3579 with 115 Rammus Shyvana i`ve faced the same problem. For me it solved enabling updates on branch options and then updating the OS. It seems that in new versions they try to update play store to the latest version before enable it. Hope it helps

closed 2023-12-02 by Dnalsi