sebanc/brunch #1892: Installation Stuck at "ChromeOS rootfs patches are being applied"

VirtualStorm1001, 2023-07-22:
Yeah, so i need ChromeOS to develop things for it and... I get stuck on "ChromeOS rootfs patches are being applied" I'm stuck here like 30 minutes.
[image]

TASvf, 2023-08-05:
Estou com o mesmo problema

ejcupcake1234alt, 2023-08-14:
on some slower systems, this stage can take 1+ hours.

elio-design, 2023-08-30:
> on some slower systems, this stage can take 1+ hours.

so did your system work ?

TASvf, 2023-09-02:
Na verdade deixei mais de 6 horas e não funcionou

TASvf, 2023-09-02:
Bom dia, tudo bem, eu esperei mais de 6 horas, e até tentei de várias
formas, instalar dentro do Linux, criei a imagem iso dentro do Windows, mas
sem sucesso, meu processador é um Intel de 4 geração, ou seja, atende os
requisitos