sebanc/brunch #1894: Sandy/Ivy Bridge CPU support

calmsacibis995, 2023-07-26:
On the README, it says that Brunch only supports Haswell CPUs and newer. But I have been able to successfuly install ChromeOS on my PC that has a Intel Core i5-3570. I have this question. The reason why 2nd/3rd gen Intel processors are not supported is because the rammus Chromebook has a Haswell processor, or it is something else, like driver support?

tejasraman, 2023-10-02:
> On the README, it says that Brunch only supports Haswell CPUs and newer. But I have been able to successfuly install ChromeOS on my PC that has a Intel Core i5-3570. I have this question. The reason [...]

Brunch actually works almost fine on Ivy Bridge (except for the Android container; that's why the `README` says it's unsupported).
`rammus` is actually Kaby Lake if you're wondering; all the actual Haswell images were made EOL in August 2019.

madhits, 2024-10-31:
> Brunch actually works almost fine on Ivy Bridge (except for the Android container; that's why the `README` says it's unsupported). `rammus` is actually Kaby Lake if you're wondering; all the actual [...]

Haswell ended with coral 126, not in 2019. If you want an updated haswell look at coral. @tejasraman I'd love your input on my other tread asking above arcvm and why kernel 6.x seems to be required except with rammus r116?