sebanc/brunch #1902: Sound Not Working on Brunch

CrazyPenguin0111, 2023-08-06:
So I recently installed chrome os + brunch on my Samsung Galaxy Book 12 from 2017. And I cannot get the sound working using wired headphones or the internal speakers. Sound works using bluetooth headphones though. Anyone got any ideas on how to fix this.

I used the latest release of brunch and version 114 for the chromos recovery image

CrazyPenguin0111, 2023-08-06:
Just for some extra info the model number is SM-W720
and it has a intel i5 7200u cpu

agrimpelhuber, 2023-08-20:
@CrazyPenguin0111 , did you play around with the various sound-related options in the Brunch setup menu?

ly9999999, 2023-09-03:
After downloading the sof firmware, refer to this method. https://github.com/sebanc/brunch/issues/1859#issuecomment-1689698613