sebanc/brunch #1904: Opening Android Apps causing Graphical Hang on ChromeOS  100 - 114

neptalactone, 2023-08-10:
Hi, it's me again with my problematic laptop.

I make this issue as follow up to my old issue post here:
https://github.com/sebanc/brunch/issues/860

So, i decided to install brunch chromeOS again with brunch 114 and chromeOS 114 with zork/ezkinil image.
I'll list out what i have tried and what works:
- I use internal_mic_fix2 to fix my microphone issue that didn't work anywhere
- I use kernel version 6.1 and kernel version 5.15
- With this image i am able to pass ChromeOS logo and get to the desktop and everything work!... except
- Everytime i open android app, it cause the graphic to hang and i have to switch to developer console (the ttys) and back to graphical console again, to make the graphic progressing and close the app.
- When I close the android app, everything back to normal again. (of course i have to switch to tty and back to close it)
- If any android app notifications appear on chrome notification, and i open the control panel thingy on the bottom right, it makes graphical hang again and i have to reboot through ttys

My Hypothesis:
- I think this have something to do because this laptop have unusual AMD iGPU + AMD dGPU combo which probably makes the android app somehow using one card that different from the system, that cause chromeos graphic to hang.
- I think the very culprit of this behavior is when chromeos introduce the android app feature that let you choose to display the android app as phone, tablet, or resizeable. Because before this features got introduce, everything works well
- I also tried FydeOS 16 & 17 version. both works wonder in this laptop except the microphone (i cannot use laptop internal microphone, but external microphone works)
- And i think the reason android implementation works on FydeOS right now, because it haven't introduce the feature that i mention on my previous point

Is there anyway to force the android container to use the same GPU with the system so it won't freak out and get the whole system experience graphical hang. I already tried to explore ChomeOS flags, and i didn't see any flags related to this. I appreciate any insight to this problem! 

If there is none of workaround available right now, i think the best possible way is to wait google found the problem itself and fix it or if they have a new chromebooks with AMD APU + AMD GPU like this.

For reference my laptop specification:
Acer Nitro 5 AN515-43
AMD Ryzen 5 3550H with Radeon Vega 8 Mobile APU
AMD RX 550X Pro
RAM 32 GB

sebanc, 2023-08-10:
I have added an experimental patch that you could try:
- Identify your dGPU pci address in "sudo lspci" (first column, in the form XX:XX.X).
- Add "remove_dgpu=0000:XX:XX.X" as an additional kernel command line parameter. (replace XX:XX.X with the pci address of your dgpu)

neptalactone, 2023-08-14:
Thanks I'll try it later !

Daubfy, 2025-01-20:
> I have added an experimental patch that you could try:
> 
> * Identify your dGPU pci address in "sudo lspci" (first column, in the form XX:XX.X).
> * Add "remove_dgpu=0000:XX:XX.X" as an additional kernel command line parameter. (replace XX:XX.X with the pci address of your dgpu)

How ?