sebanc/brunch #1905: iptsd failed to launch due to missing libfmt.so.9

Cyanoxygen, 2023-08-10:
Hi,
I have a Surface Pro 7 Plus which uses ITHC touchscreen. Touchscreen worked after enabling `ithc_touchscreen` option, but only one finger is registered and the Surface Pen is not usable.

After a bit of digging I found that the user space daemon `iptsd` is not launched, and when I manually run it, it raises a missing library error:

```
crosh $ shell
localhost $ sudo -s
localhost / iptsd
iptsd: error while loading shared libraries: libfmt.so.9: cannot open shared object file: No such file or directory
```

A `find` showed it does not exist, maybe it is not installed into the framework.

----

Brunch framework: r114
ChromeOS version: 114.0.5735.239
Kernel: Brunch 6.1
Options: ithc_touchscreen

sebanc, 2023-08-11:
Could you try this build ?
https://github.com/sebanc/brunch-unstable/releases/tag/r115-unstable-20230810

Cyanoxygen, 2023-08-11:
Thanks! I am going to install this version of framework then, please wait for my test result.

Cyanoxygen, 2023-08-11:
Hi,
The iptsd is up and running, and multitouch works fine. Surface Pen worked too.

sebanc, 2023-08-11:
Thanks for your feedback !

Cyanoxygen, 2023-08-11:
I think it is better to close this issue after the next stable release.

closed 2023-08-16 by Cyanoxygen