sebanc/brunch #1909: android not opening

ale-abo, 2023-08-18:
android app are not opening, i click the icon and start loading forever

agrimpelhuber, 2023-08-20:
@ale-abo, do you expect anyone to answer an issue as unspecific like this?

Well, maybe good old me: Had similar issues. Sometimes, it's just the android subsystem taking longer than expected in the background to start up. Be patient.

If that does not help, google something like "brunch android VM not started" or something like that. There's plenty of information on problems like this, and how to diagnose them.

For me, it was probably the fact that I updated ChromeOS (via the official way - settings) to a version that wasn't yet supported by brunch. There are plenty of warnings that this is risky, but what do I care? ;-)

I fixed it with a combination of (not necessarily in this order):
- disable Terminal, if enabled
- disable Play Store (Android)
- update Brunch via PWA to the newest available version (unstable)
- update ChromeOS via PWA to the corresponding recovery version
- enable Play Store again

Let me know if it worked.

agrimpelhuber, 2023-08-20:
and if you had searched, you had found [this](/sebanc/brunch/issues/1889)

ale-abo, 2023-08-23:
ah thanks I have searched but I found only old guides for brunch 80 and 74 that haven't worked and yes I tried wait all night and nothing happened, I tryed power wash and disabling and enabling the play store but thanks

closed 2023-08-27 by ale-abo