sebanc/brunch #1911: Confusing Instructions

johnaweiss, 2023-08-25:
Hi, 

This is very confusing:

- on the page called "Install Brunch with Windows", the instructions say "Log into ChromeOS"
  - How does logging into ChromeOS install Bruch with Windows?
- Under the section "Install Brunch", it says "sudo chromeos-install -dst /dev/disk". 
  - How does "chromeos-install" install Brunch?

https://github.com/sebanc/brunch/blob/main/install-with-windows.md

Can someone please explain?

Thanks

ZProLegend007, 2023-08-25:
I believe you need to make a Brunch USB and boot off that to install. I believe once you have made that and booted off the usb, you follow the instructions to access the linux shell from crosh in chromeOS to then run a built in script to install brunch on the main device.

johnaweiss, 2023-08-25:
> I believe you need to make a Brunch USB

Thx. Do you mean, make a Linux USB, with brunch in a folder?
Have you done this?

thx

ZProLegend007, 2023-08-26:
You need to follow the instructions clearly stated on the install pages to make the usb. You must first make the chromeos .img file then use a tool such as etcher or rufus to create a bootable usb of brunch. You can then follow the install instructions from the usb.

closed 2025-12-18 by github-actions[bot]