sebanc/brunch #1913: Asus FA506ICB - F15

eldertek, 2023-08-28:
ASUSTeK COMPUTER INC. ASUS TUF Gaming F15
CPU: AMD Ryzen 5 4600H with Radeon Graphics
RAM: 19423 MB
Storage: SAMSUNG MZVLQ512HBLU-00B00               476 GB
Brunch: Brunch r115 20230814
Recovery: zork 115 (build 15474.84.0)
- Does the Android Subsystem (Android apps) run?
[Yes] with android_init_fix
- Is Linux container install option available?
[Yes] 
- Does Linux container install?
[Yes]
- Does Linux terminal run?
[Yes]
- Is Bluetooth working?
[No] Can pair with bt_console, but cannot connect to the chromebook. I think that's the reason why app streaming don't work with my phone.
- Is WiFi working?
[Yes]
- Is the camera working?
[Yes] with internal_mic_fix2
- Is microphone working?
[Yes]
- Does the touchscreen work?
[Not Avaliable]
- Does trackpad and gestures work?
[Yes]

Others options:
- pwa
- acpi_power_button

ZProLegend007, 2023-08-29:
Try having a look through previous github issues relating to bluetooth, you may be able to find a solution there :)

eldertek, 2023-08-29:
> Try having a look through previous github issues relating to bluetooth, you may be able to find a solution there :) Also I believe chromeOS doesn't support app streaming... I read that somewhere in the brunch wiki...

Thank you ! I sucessfully paired to my phone using bluetoothctl

Perhaps even if smartphone paired there is not bluetooth devices in chrome os settings

And no app streaming working, I don't see any reference in wiki, maybe need to find more ...

closed 2025-12-18 by github-actions[bot]