sebanc/brunch #1914: Running Brunch on Macbooks - Fixes and Issues.

ZProLegend007, 2023-09-04:
Hello there,

Previously I had made an issue about specific problems while running brunch on my Macbook, there are still some issues but I will show you how you can get a decently running setup if you are having issues on your T2 Mac with brunch.

A T2 Mac is any Mac made during or after 2018 that  _isn’t_ Apple Silicon (M).

My model is a Macbook Pro 2018 T2 with Touchbar.

### What works and what doesn't.

- [ ]  Keyboard
- [ ]  Keyboard Backlight
- [ ]  Brightness
- [ ]  Ethernet
- [ ]  USB/Thunderbolt - (Working properly after set kernel params.)
- [ ]  Touchpad (working on brunch/chromeOS v103 and lower confirmed. | Newer versions not working. (Confirmed on v110+))
- [ ]  Battery - (Battery life is significantly lower than other OSs but works fine.)
- [ ]  Touchbar keyboard backlight controls - (Dependent on Touchbar)
- [ ]  Power button (After selecting acpi_power_button in brunch settings.)
- [ ]  Bluetooth audio (After Wi-Fi patch, flawless.)
- [ ]  Bluetooth (After Wi-Fi patch.)
- [ ]  Wi-Fi (After Wi-Fi patch.)
- [ ]  Android/Play Store - (Very occasionally can have some graphical issues.)
- [ ]  Touchbar (Partial/Depends on version.)
- [ ]  3.5mm Jack Audio | Wired Headphones - (Partial - Not automatically detected. Partially work by using commands below.)
- [ ]  Internal Speakers audio 
- [ ]  Suspend/Sleep - (Causes shutdown.)
- [ ]  Ambient Light Sensor - (For this release at least.)

What can sometimes not work (can usually be fixed by reinstalling brunch);

- Shutdown - (Doesn't properly halt, causes device to stay on while everything else seems off (can tell by caps lock key light).)

### What's the best way to run brunch on a Mac?

The best way for me was to use **brunch r103** with chromeOS recovery image **rammus v105**.

My setup is using rEFInd for my other operating systems and using rEFInd to boot grub which boots brunch. Brunch itself is installed in my bootcamp partition in a folder.

(Good thing to note is that grub2win **does not** work on bootcamp as Windows doesn't have access to the efi partition, so you need to install it in a different way, and grub seems to be the only bootloader that supports the way that brunch boots.) 

**Keep in mind that I have only tested this on my own laptop, these instructions may vary between devices and depends on your specific setup.**

**IMPORTANT: Everytime you rebuild the rootfs you will need to redo this process from steps 5-7. I have created a script to run all those commands for me everytime I need to enter them. I recommend you do the same. Make a text document that has all the commands that needs to be run on a new line for each and then save it as something.sh in your downloads. Make sure your working Wi-Fi firmware file is also in your downloads folder.**

What works straight away is... Nothing. You can boot but nothing except an external keyboard will work.

1.  You need to go to brunch settings. On most Macs the built in keyboard won't work with this so you will need an external keyboard.  You need to set the kernel to 'macbook'. Then on the next page select 'acpi_power_button'. On the next page you don't need to select anything. The page after where it asks for kernel parameters, enter this: "intel_iommu=on iommu=pt pcie_ports=compat". This will let USB/Thunderbolt/Ethernet devices work when plugged in, without this they may work if plugged in at boot but not after. Press enter, then select your bootsplash and reboot.

2.  Now we have to get you some Wi-Fi firmware. You can do this in a few ways but the way I will show you requires you to go into MacOS. So boot in to MacOS and open the terminal, then run this script “https://wiki.t2linux.org/tools/firmware.sh”. 

`sudo bash ~/Downloads/firmware.sh`

This will extract your Wi-Fi firmware and place it in your EFI partition, that's not exactly what we want but it works. Next you will need to mount your EFI partition in the terminal by running

```
mkdir ~/efi 
sudo mount -t msdos /dev/disk0s1  ~/efi
```

Now if you open finder and go to the EFI location, you should see a tarball of your firmware somewhere around there. Once you have found it, move it to your downloads. Then extract the firmware by double clicking it. Now you will need a USB Drive. Plug your drive in and go to finder to copy the wifi folder under firmware to your drive.

5.  Great! Now when you boot up into brunch next... It will crash and reboot... This is normal, turns out that there is something wrong with the als driver in this release (Ambient Light Sensor), but lucky for us we can disable the touchbar and als by doing a force shutdown. (If you are booting an OS that isn't Bootcamp or MacOS and you do a force shutdown then the touchbar will not show on the next boot.) So do a force shutdown by booting into an OS or sitting on your bootloader and holding the power button until the screen turns off. Then wait a couple seconds and turn your Mac back on. When you boot into brunch now, it shouldn't crash. If it does then try again. Now you should have your keyboard, backlight and touchpad working. Touchbar won't be working but we can fix that later.

6.  Now you will need to plug your Mac into Ethernet as we will have no Wi-Fi in chromeOS. Once you have done that then you can either log in or use it as a guest. Once you are at the home screen press Ctrl+Option+T to open up the crosh terminal. Once you are in the crosh terminal type 'shell' to enter the linux shell then sudo su. Now open the files app and copy your files over from the USB to the Downloads folder. Going back to terminal theres a couple of things you need to do. Firstly you need to run this: 

`rm /lib/modules/5.10.120-macbook-brunch-sebanc/kernel/drivers/misc/apple/ib/apple-ib-als.ko `

This will remove the Ambient Light Sensor driver and stop the crashes on boot. Next you need to run 

[... 59 lines cut ...]
Now make sure in the chromeOS ui that you have set the output to be your bluetooth device. Once you’ve done that and can confirm audio works from that then enter this command:

`cras_test_client –select-output 9:0`

Your audio should now be coming out of your 3.5mm audio jack. This command tells cras (ChRomeos Audio Server) to output to “Apple T2 Audio: Codec Output:1,2” which seems to be the jack!

Things that could be improved:

-  Battery life
-  Android/Play Store Graphical anomalies

Battery life is most likely caused by a lack of power saving settings enabled. The brightness can also cause the estimated battery life to go from 6h to 2h from off to max brightness. High cpu usage can tend to half the battery life.

###Issues that need to be fixed.

-  **Internal/Speaker Audio**
-  Sleep/Suspend
-  Touchpad in latest release
-  Wi-Fi in latest release

Info for devs on Internal Speaker Audio.

I did a lot of research and experimentation to troubleshoot audio but I have determined that I cannot fix this alone. From what I can tell, cras controls all the audio playback. When any playback is attempted to be sent to the internal speakers, cras crashes. The logs say stream error. Youtube says audio renderer error, please restart your computer. I hope someone can help me come up with a solution. It has been the one issue other than Wi-Fi that has been the same across all the tested releases of brunch and versions of chromeOS. If anyone wants any more info feel free to ask.

Any feedback or help is greatly appreciated.

blackxfiied, 2023-09-04:
im gay

DamianNiolet, 2023-10-09:
OK...Using 13" MBPro from 2020 (16.2 or A2251):

- I tried r107/v108 and it wouldn't even get past the "Loading Brunch framework" screen.  I left it for 30 minutes and nothing.  Note: "MacBook" on the first setting page.

- Rather than jump straight to r117 (because of issues I'll describe next), I tried r114/v114.  Note: "Macbook t2" on first settings page.

- [x] Keyboard
- [ ]  Keyboard Backlight - (Appeared on Touchbar, but didn't actually light up)
- [x]  Brightness
- [x]  Ethernet
- [x]  USB/Thunderbolt - (Working properly after set kernel params.)
- [ ]  Touchpad
- [ ]  Battery - (set acpi parameter, but no indicators in ChromeOS)
- [ ]  Power button (set acpi parameter, but power button did nothing)
- [ ]  Bluetooth audio (didn't bother trying since, no audio output)
- [ ]  Bluetooth (no go)
- [ ]  Wi-Fi (possible, didn't try just yet)
- [ ]  Android/Play Store - (it started updating, but didn't test it)
- [ ]  Touchbar (comes on, but backlit keyboard doesn't change)
- [ ]  3.5mm Jack Audio | Wired Headphones - (didn't bother trying)
- [ ]  Internal Speakers audio (no)
- [ ]  Suspend/Sleep - (didn't try)
- [ ]  Ambient Light Sensor (didn't cause crashes, didn't test thoroughly)
- [x] Camera
- [x] crosh - shell - sudo

- I initially tried r117/v117 before going back to r107/v108.  Note: "MacBook t2" on first settings page.  Had all the same issues as above; however, crosh would not go to shell, so could not attempt any sudo commands.  Maybe it was my error on my first go-around installing brunch, so I'll try it again later.  But wanted to take @ZProLegend007 guidance and focus on consistencies for stable versions first.

Will keep on trying other variations according to @ZProLegend007 plan, but wanted to report this effort thus far.

ZProLegend007, 2023-10-09:
Hmm, that is rather interesting. Thank you for your input. Regarding the crosh issue - I can load crosh and shell while testing on my MacBook, what happens when you type in `shell` ?

sebanc, 2023-10-12:
You seem to be putting a lot of energy into this but unfortunately the way you proceed to debug things is not really helpful for me.

The kernel patches related to macbook-t2 are available here:
https://github.com/sebanc/brunch-unstable/tree/continuous-integration/kernel-patches/chromebook-6.1/others
They are taken from here:
https://github.com/t2linux/linux-t2-patches/tree/6.1

The kernel configs for macbook-t2 are taken from here:
https://github.com/t2linux/linux-t2-patches/blob/6.1/extra_config
And added at the end of the brunch configs here:
https://github.com/sebanc/brunch-unstable/tree/continuous-integration/kernel-patches/brunch_configs

Regarding touchpad, the CONFIG_MOUSE_BCM5974 is not listed in https://github.com/t2linux/linux-t2-patches/blob/6.1/extra_config so it was not enabled. I just enabled it in the latest brunch-unstable but I would have appreciated a pull request.

Regarding battery, it is probably a missing config as well but I don't know which one.

Regarding wifi, all the 6.1 patches from https://github.com/t2linux/linux-t2-patches/tree/6.1 are applied so I guess the issue comes from the firmwares. Please attach the correct firmwares as I don't have this device and cannot extract them.

Regarding audio, I guess a specific ucm config file would be needed for sound to work but I cannot do anything about it.

ZProLegend007, 2023-10-12:
Hi sebanc. Thank you for your response. I'm not a coder myself I just play around and investigate until I find a way to make things work. I'm just trying to lay things out in the way that makes most sense from what I can tell. I also haven't used pull requests yet but I'll take a look and try to start using them.

Regarding Wi-Fi, [this patch](https://github.com/t2linux/linux-t2-patches/blob/6.1/5001-Revert-wifi-brcmfmac-cfg80211-Pass-the-PMK-in-binary.patch) was removed in r114 and began causing the error then. There is firmware though and the firmware does work (thank you).

ZProLegend007, 2023-10-12:
I will be applying fixes to a fork of this project.

DamianNiolet, 2023-10-12:
Hello, @sebanc!  Just wanted to say, I appreciate your work, your assistance, and your feedback.  Looking forward to a fully operational ChromeOS on my 2020 t2 macbook pro!

DamianNiolet, 2023-10-12:
@ZProLegend007, bluetooth does not work in my r117.  Does it work "out for the box" for you?  Wifi works, but doesn't connect to my WPA network, as expected.

ZProLegend007, 2023-10-12:
I believe so, I'll test in a moment.

sebanc, 2023-10-12:
Are you sure the firmwares I added are correct ?

I kind of blindly added them based on what I could find on the macbook-t2 github but instructions were not very clear.

Moreover, the patch that you mention here https://github.com/sebanc/brunch/issues/1942 and that I removed in r114 dates back from when the macbook-t2 kernel was 5.15, with kernel 6.1 (current) the patch is applied again.

Incorrect firmwares would explain bluetooth and wifi issues but I would need the right ones.

sebanc, 2023-10-12:
Btw, sorry did not mean to be harsh this morning or for you to close the issue, I just meant I don't understand those issues and I would need help to resolve them.

ZProLegend007, 2023-10-13:
The firmwares are indeed correct. Thank you. It's alright, I figured you were busy with other things and macbooks weren't too much of a priority so I forked brunch-unstable hoping to be able to apply fixes myself... Turns out it takes a looooooooooooooooooooong time to even just build the kernels. I have my regrets 😅.

ZProLegend007, 2023-10-13:
My understanding and explanation of that patch was quite off I believe. I see now that [this patch](https://github.com/sebanc/brunch-unstable/blob/f79452852ac9e294cc7c6b8ea22df41381519423/kernel-patches/chromebook-6.1/others/5001-Revert-wifi-brcmfmac-cfg80211-Pass-the-PMK-in-binary.patch) reverts a [patch ](https://github.com/sebanc/brunch-unstable/blob/02e00fc040f3222b6cd1009533e12a28d6b0f5b1/kernel-patches/chromebook-5.15/others/8022-brcmfmac-cfg80211-Pass-the-PMK-in-binary-instead-of-.old).

ZProLegend007, 2023-10-13:
Regarding audio, does this help? https://gist.github.com/MCMrARM/c357291e4e5c18894bea10665dcebffb

ZProLegend007, 2023-10-13:
Regarding battery config, what about `CONFIG_GENERIC_ADC_BATTERY=m`? This isn't in the latest brunch_configs file.

ZProLegend007, 2023-10-13:
> @ZProLegend007, bluetooth does not work in my r117. Does it work "out for the box" for you? Wifi works, but doesn't connect to my WPA network, as expected.

I did just test, bluetooth doesn't turn on at all which is odd. I'll see if I can find any errors or if I can use `bt_console` in crosh.

sebanc, 2023-10-13:
> Regarding audio, does this help? https://gist.github.com/MCMrARM/c357291e4e5c18894bea10665dcebffb#file-applet2-conf
> 
> I assume that's a config file for the speakers?

The pulseaudio part of those configs will not work but maybe the alsa part would be sufficient. Could you try to manually add the file /usr/share/alsa/cards/AppleT2.conf and see if it helps with sound ?

sebanc, 2023-10-13:
> CONFIG_GENERIC_ADC_BATTERY

Thanks it looks good ! I am going to add it

ZProLegend007, 2023-10-13:
> > Regarding audio, does this help? https://gist.github.com/MCMrARM/c357291e4e5c18894bea10665dcebffb#file-applet2-conf
> > I assume that's a config file for the speakers?
> 
> The pulseaudio part of those configs will not work but maybe the alsa part would be sufficient. Could you try to manually add the file /usr/share/alsa/cards/AppleT2.conf to your brunch install and see if it helps with sound ?

It didn't seem to work, but checking my other linux OS, I don't see it there either so maybe theres something else going on?

sebanc, 2023-10-13:
Thanks for confirming that the wifi firmwares are correct, unfortunately the kernel code changes quite often and it is not really possible to use patches that were not made for the version you are working with.

Could you please post a full dmesg ?

ZProLegend007, 2023-10-13:
Ah yes alright.

[dmesg.txt](https://github.com/sebanc/brunch/files/12890847/dmesg.txt)

ZProLegend007, 2023-10-13:
Testing the latest brunch-unstable shows no change, battery still not detected. I'll keep looking for configs. It is good that applesmc was added back though.

ZProLegend007, 2023-10-13:
I ran through all of the configs I could find in brunch r107 (where the battery was detected and bluetooth worked.) that affect power management and battery and one relating to bluetooth. There are a lot... Perhaps these could present a fix?

--- Power related ---
CONFIG_MFD_TPS68470=y
CONFIG_TPS68470_PMIC_OPREGION=y
CONFIG_GPIO_TPS68470=y
CONFIG_INTEL_SOC_PMIC=y
CONFIG_INTEL_SOC_PMIC_BXTWC=y
CONFIG_INTEL_SOC_PMIC_CHTWC=y
CONFIG_INTEL_SOC_PMIC_CHTDC_TI=y
CONFIG_INTEL_SOC_PMIC_MRFLD=y
CONFIG_PMIC_OPREGION=y
CONFIG_BYTCRC_PMIC_OPREGION=y
CONFIG_CHTCRC_PMIC_OPREGION=y
CONFIG_XPOWER_PMIC_OPREGION=y
CONFIG_BXT_WC_PMIC_OPREGION=y
CONFIG_CHT_WC_PMIC_OPREGION=y
CONFIG_CHT_DC_TI_PMIC_OPREGION=y
CONFIG_CRC_PMIC_OPREGION=y
CONFIG_PMIC_ADP5520=y
CONFIG_INTEL_SOC_PMIC_BXTWC=m
CONFIG_INTEL_SOC_PMIC_CHTDC_TI=m
CONFIG_WM831X_POWER=m
CONFIG_WM8350_POWER=m
CONFIG_MAX8925_POWER=m
CONFIG_PDA_POWER=m
CONFIG_DPTF_POWER=m

--- Bluetooth related ---
CONFIG_BT_RTL=m
CONFIG_BT_QCA=m
CONFIG_BT=y
CONFIG_BT_RFCOMM=y
CONFIG_BT_HS=y
CONFIG_BT_INTEL=m

sebanc, 2023-10-14:
Actually there were many errors in your dmesg related to the apple-bce module, it seems that the linux-t2 patches for kernel 6.1 are not stable.

I rebased the kernel-macbook-t2 to 5.15 in the latest brunch-unstable release.

Let me know what works.

ZProLegend007, 2023-10-14:
Ah interesting, thank you! I'll test it out asap and see how it goes.

ZProLegend007, 2023-10-14:
I'm just waiting to get another externel keyboard to switch kernels (I'm out at the moment.) That does make me think though, and I don't know if this is possible or worth it to do, but built-in keyboard support for brunch settings?

ZProLegend007, 2023-10-14:
Update: The new release doesn't work :(

Battery works now.

No keyboard, touchbar or touchpad.

Wi-Fi is still giving that error.

sebanc, 2023-10-15:
> I'm just waiting to get another externel keyboard to switch kernels (I'm out at the moment.) That does make me think though, and I don't know if this is possible or worth it to do, but built-in keyboard support for brunch settings?

Not directly but I might have a workaround, does the keyboard work in grub ?

Could you try the latest brunch-unstable release ?

ZProLegend007, 2023-10-15:
Keyboard does works in grub.

--I'll edit in what works on the new release here--

ZProLegend007, 2023-10-16:
Or if the 5.10 rebase doesn't work try 5.4. I believe that was the most stable release. (Or even 4.4.)

ZProLegend007, 2023-10-16:
I'm testing the latest brunch-unstable now.

ZProLegend007, 2023-10-16:
The 5.10 rebase also has no keyboard, touchbar or touchpad. Wi-Fi is still broken with that error **when adding my own firmware because the included firmware was removed.** I think we should try troubleshoot kernel 6.1. Pls try those configs and also adding back the firmware. Also maybe try not adding [this patch](https://github.com/sebanc/brunch-unstable/blob/f79452852ac9e294cc7c6b8ea22df41381519423/kernel-patches/chromebook-6.1/others/5001-Revert-wifi-brcmfmac-cfg80211-Pass-the-PMK-in-binary.patch) to 6.1 and seeing what happens.

sebanc, 2023-10-16:
What I did was just to re-apply the kernel patches and firmwares as they were in brunch r107 (that was supposedly working). I am very confused as to why that does not work anymore.
Unfortunately, I am pretty sure that the configs you mention will have no effect.

The linux-t2 patches regardless of the kernel version do not seem to play well with chromeos and I cannot debug that as I do not have such device.

I let it to someone else to work on this and to submit the adequate pull requests.

ZProLegend007, 2023-10-16:
Hmm alright. Thank you for your help sebanc :)

DamianNiolet, 2023-10-17:
> The linux-t2 patches regardless of the kernel version do not seem to play well with chromeos and I cannot debug that as I do not have such device.

@sebanc, I know this is an unconventional proposal, but I will gladly send you a model of my MacBook Pro.  You don't even have to send it back.  Wish you could DM people on here...

ZProLegend007, 2023-10-17:
I also believe I have found a fix for the keyboard, touchpad and touchbar not working in the last brunch-unstable. You removed these configs which are necessary for some major things to work.

CONFIG_APPLE_BCE=m
CONFIG_APPLE_GMUX=m
CONFIG_BRCMFMAC=m
CONFIG_BT_BCM=m
CONFIG_BT_HCIBCM4377=m
CONFIG_BT_HCIUART_BCM=y
CONFIG_BT_HCIUART=m
CONFIG_HID_APPLE=m
CONFIG_HID_APPLE_IBRIDGE=m
CONFIG_HID_APPLE_MAGIC_BACKLIGHT=m
CONFIG_HID_APPLE_TOUCHBAR=m
CONFIG_HID_SENSOR_ALS=m
CONFIG_MOUSE_BCM5974=m
CONFIG_SENSORS_APPLESMC=m
CONFIG_SND_PCM=m
CONFIG_STAGING=y

Could you maybe release one more brunch-unstable with these added back?

sebanc, 2023-10-17:
> CONFIG_BT_BCM

Those configs (aside from the touchpad one that I forgot) were not necessary with the kernel patches used in brunch r107. I will re-add the patches and firmwares to the 6.1 kernel so that touchpad and touchbar work and maybe someone can take it from there.

sebanc, 2023-10-17:
> @sebanc, I know this is an unconventional proposal, but I will gladly send you a model of my MacBook Pro. You don't even have to send it back. Wish you could DM people on here...

That's a very nice offer, however actually even if I had this device I am not a kernel developer, I just apply the patches from the t2-linux GitHub so the chances that I could make this work are very low.

ZProLegend007, 2023-10-17:
> Not directly but I might have a workaround, does the keyboard work in grub ?

Could you also try adding that workaround? That would save me a lot of time, thanks.

sebanc, 2023-10-18:
> Could you also try adding that workaround? That would save me a lot of time, thanks.

I have launched the build of the brunch-unstable release with macbook patches reverted to kernel 6.1, it should be out in a few hours.

The workaround to have keyboard support in brunch-settings would be:
- highlight the brunch-settings boot option in grub.
- Press "e"
- Replace "/kernel" by "/kernel-macbook"
- press CTRL+X to boot into the brunch-settings menu

ZProLegend007, 2023-10-18:
Thank you, I have just figured out how to properly build a fork of brunch-unstable so I will go ahead with attempting fixes myself, thank you for your help. I will add pull requests for anything I find. 

I'll try altering grub settings permanently and seeing if that works. Thanks!

ZProLegend007, 2023-10-18:
Hey @sebanc !! Battery issue is fixed. I added those PMIC configs and the battery works again. I was going to submit a pull request but it's a simple fix.

ZProLegend007, 2023-10-19:
> * highlight the brunch-settings boot option in grub.
> * Press "e"
> * Replace "/kernel" by "/kernel-macbook" in the config
> * press CTRL+X to boot into the brunch-settings menu

I have tried this and also tried adding it permanently to grub. It loads the kernel but there is still no keyboard.

ZProLegend007, 2023-10-19:
WIFI FIXED!!!! It's a kernel parameter!! If you can add this parameter for the MacBooks then it should work.

`brcmfmac.feature_disable=0x82000`

DamianNiolet, 2023-10-19:
You two are awesome!  I'm writing this from my office in brunch/ChromeOS!

I've written out the complete guide for getting rEFInd to boot brunch/ChromeOS directly.  Is this something useful to others, so much so you'd like me to post somewhere?

ZProLegend007, 2023-10-19:
There were some configs (settings for when the kernels are being built) that were missing and so the power management chip wasn't being detected. 

A guide may need to be added but let's get as much fixed as we can and then make one. Glad to help!!

sebanc, 2023-10-20:
Thanks for the debugging.

I should be able to add the wifi fix but before I launch a new build I would like to confirm that:
- the battery is fixed in the latest brunch unstable build
- the workaround for keyboard in brunch-settings works in the latest brunch unstable build
Could you test this ?

ZProLegend007, 2023-10-20:
Will do, I will test that in a bit.

ZProLegend007, 2023-10-20:
Battery is indeed fixed, and the workaround works!

sebanc, 2023-10-20:
> Will do, I will test that in a bit. If you can add that parameter could you also add `intel_iommu=on iommu=pt pcie_ports=compat` as well? They are necessary for USB and other things to function properly.

I have just launched the new brunch-unstable build that contains the wifi fix.
However, I cannot add those kernel parameters directly as they will cause issues with other devices (especially iommu).

Regarding bluetooth, there is no error in the log, as a reminder chromeos cannot connect to phones via bluetooth due to the specific phone hub.

Regarding als, I guess it works but not as you would expect:
https://chromium.googlesource.com/chromiumos/platform2/+/master/power_manager/docs/screen_brightness.md

For Suspend and Audio I have no idea.

ZProLegend007, 2023-10-21:
Checking logs and comparing them to other logs on working linux, Bluetooth and all it's services initialize but there are no `Bluetooth: hci0: blah blah blah` messages and it doesn't even try to load firmware. Usually these messages come after brcmfmac is loaded, which makes sense as its a dual-mode chip that does both Bluetooth and Wi-Fi. But there's nothing which is odd.

DamianNiolet, 2023-10-21:
I'd like to try the latest unstable version as well, but I'm a bit confused.  Do I still need to follow your install guide @ZProLegend007, or will all those adjust (selecting/writing in parameters) already be built in?  Am I going to have to do the "press e" and change kernels thing?

ZProLegend007, 2023-10-21:
So the install guide was for the old versions. Now just install as you would normally but with some new things. When you get brunch installed and you have the grub script, make sure you change the kernel for brunch-settings to the MacBook kernel (basically in grub-customiser or the grub.cfg file in the brunch settings entry you change the /kernel to /kernel-macbook so you can change things with the built in keyboard.) Then everything else should work. Otherwise you could use my fork which only has the MacBook kernel and some other customisation (I got a bit carried away hehe) and you don't have to do anything other than install it the normal way. Everything is already configured except the default MacBook kernel parameters (iommu e.t.c). You should probably run through the brunch settings on that too, I have a few things turned on by default.

ZProLegend007, 2023-10-23:
More info: 

Bluetooth is on the same card as Wi-Fi, they coexist. Usually in dmesg the Wi-Fi firmware will load and then that will trigger the Bluetooth firmware to load as well. Instead the Bluetooth services initialize as normal but there is no hci0 anywhere and no firmware loading for Bluetooth. I'm almost stumped here.

Audio I believe is setup correctly. Everything I can find is the same compared to my other linux distro. I believe cras (ChRomeos Audio Server) crashes because of... something... and I have no idea what. The speakers are detected correctly and show, the mic is detected and does work too.

ALS does work, but I find it is too sensitive. It's always too bright, to the point where if it is dim and it sees the display reflection from my hand it can go all the way up to max brightness even in a dark room.

Then obviously the shutdown on sleep which should be simple to fix once I start looking into that more deeply.

Any ideas?

DamianNiolet, 2023-10-24:
Great work, @ZProLegend007!  Love that I don't have to change parameters.

Have you noticed the touchpad randomly jumping to the bottom left corner though?

I see a new release just minutes ago.  Keep up the good work!  You'll figure it all out.  :)

ZProLegend007, 2023-10-24:
Ahh interesting, that doesn't happen on my device. Does that happen on any other OS or distro?

DamianNiolet, 2023-10-24:
> Ahh interesting, that doesn't happen on my device. Does that happen on any other OS or distro?

Nah, just with the ChromeOS partition.  Doesn't happen when I use the external mouse, so something about the 2020 trackpad is slightly different.  Not a big deal right now though.

ZProLegend007, 2023-10-24:
Ahh interesting, I'll do some research. Does this happen on brunch-unstable?

DamianNiolet, 2023-10-25:
I've yet to try your latest, as I need a stable version to try something else.  I'm trying to virtualize the Brunch/ChromeOS partition.  I've gotten it as far as the ChromeOS logo, but then it hangs.  Verbose shows endless "tpm timeout errors."  I know the chances are slim, but I've got my reasons.  lol.

ZProLegend007, 2023-10-25:
Haha, well that is very interesting! Chromebooks have TPMs so it needs to be emulated or faked on brunch, I guess that doesn't play well with virtualization... Is this in windows? What software are you using to virtualize it?

[22 more replies]