sebanc/brunch #1921: ChromeOS boot issue

SSP6904, 2023-09-17:
Hello all!

I'm using chromeOS on my Dell Latitude E6540 with the latest release of rammus.

However, the newer releases seam to make it not boot after the brunch screen. It first gets stuck on the ChromeOS logo, but then dims the display and turns off after about ten minutes. I have no idea why it does this, since the version I last used was 108.

My GPU is compatible with the system, and it seams that the release I'm attempting to install which is 115 (as of writing this) isn't letting it boot due to the GPU inside of my laptop.

This version was working on my Lenovo laptop and this Dell latitude a while back, but seams to fail on me.
Can someone help me out?

Thank you!

ZProLegend007, 2023-09-17:
I notice your model includes an intel iGPU and a discrete AMD GPU, your discrete GPU may be causing the issue here, is there any option to disable that in your bios?

mesidex, 2023-09-17:
I had the same issue except the black screen was caused because my monitor was receiving output via DisplayPort (from the GPU). Only after I had switched to HDMI did my monitor display output. I can't 100 % guarantee it's the same case as mine though

SSP6904, 2023-09-17:
I do not have that to disable my AMD GPU for the laptop in the BIOS.

I believe their might be a command for brunch for disabling that.

Other then that, I'm not sure.

ZProLegend007, 2023-09-18:
Try checking some of the older issues, they may contain instructions. Otherwise change around some graphical brunch settings and see if that changes anything.

SSP6904, 2023-09-18:
I have found an issue related to this one. #1481 

There is a way to disable the dGPU for brunch. There are a few kernel commands. I'll try them to see if that changes anything.

`fbcon=map:1`

`amdgpu.dc=0`

`amdgpu.dc=1`

SSP6904, 2023-09-24:
Came back, and none of my commands are working.

My question is still wondering about disabling the dGPU completely without the bios.

Does that work for this framework, @sebanc?

ZProLegend007, 2023-09-26:
Maybe try fbcon=map:0 ?

SSP6904, 2023-09-27:
I will test some of these commands on my Kubuntu laptop and see if it will disable my AMD card.

If they work, I will test it again on Brunch.

SSP6904, 2023-10-02:
The OS still refuses to load with the kernel command in.

I still don't know why the latest versions are not compatible with my AMD card..

ZProLegend007, 2023-10-07:
Hmm, that is unfortunate. Have you tried the brunch-unstable builds?

SSP6904, 2023-10-08:
Didn't work with the unstable build.

It still goes into the black screen without the setup prompt.

neptalactone, 2023-11-01:
Hi, @SSP6904,

Sebanc already try making an experimental custom kernel parameter to disable dedicated GPU that might causing problem.

Try to identify your dedicated GPU pci addresses with lspci command with sudo if required. you can do it from chromeOS tty, or using existing linux OS, or using WSL (idk if that will work). make note of the address, it will have a format like this: XX:XX.X (in the first column of the command).

Then enter to brunch settings either from cros tty or grub. in the additional kernel parameter, you can add "remove_dgpu=0000:XX:XX.X" with the XX:XX.X being changed with your dedicated GPU pci address that you take notes before.

You can refer to my issue post that @sebanc replied with this workaround in #1904

SSP6904, 2023-11-02:
That didn't work either. I did the lspci command, added my kernal line from the grub settings.

SSP6904, 2023-11-02:
The latest versions don't work for my laptop. Right now, 114 is working right now, but with problems I've yet to fix.

For now, I'll close this issue, and look into it more soon.

closed 2023-11-02 by SSP6904

SSP6904, 2023-11-02:
Do I have add the kernel commands with the grub settings or edit it manually?

I thought you have to edit the grub config to add commands..

SSP6904, 2023-11-02:
I fixed it! I managed to edit the grub.cfg in the boot partition and added the suggested command to the line.

ZProLegend007, 2023-11-03:
Anything that adds kernel parameters should work. Even brunch-settings.

abbasabidi85, 2024-02-25:
I added kernel parameters through brunch settings but it didn't work. I'm trying using v121 latest release, I'll try 114 rammus, i also have amd dgpu with i5 7th gen cpu.

ZProLegend007, 2024-02-28:
The parameters might not be effective for this issue... I can't really help any further but I wish you all the best with finding a solution!

abbasabidi85, 2024-02-28:
Brunch please add disable gpu flags like opencore ecosystem has for booting macOS

abbasabidi85, 2024-02-28:
> The parameters might not be effective for this issue... I can't really help any further but I wish you all the best with finding a solution!

I successfully booted into chromeOS the issue is resolved by adding `remove_dgpu=0000:XX:XX.X` in kernel command line parameters replacing XX:XX.X with my dgpu pcie address.

SSP6904, 2025-07-15:
Coming back to this issue after some time, I found something in the `dmesg` logs that would mostly likely be the cause for the system not booting into the setup program. I've attached a portion of it because of a few error messages it posted right before it tries to load the setup program.

```
[    7.949838] Setting dangerous option enable_fbc - tainting kernel
[    7.950480] Setting dangerous option enable_psr - tainting kernel
[    7.951403] pci 0000:00:02.0: attach allowed to drvr i915 [internal device]
[    7.952197] i915 0000:00:02.0: [drm] Found HASWELL (device ID 0416) display version 7.00 stepping N/A
[    7.953283] Console: switching to colour dummy device 80x25
[    7.953325] i915 0000:00:02.0: vgaarb: deactivate vga console
[    7.955882] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    7.992683] drm_get_panel_orientation_quirk called with width=1920 height=1080
[    8.014315] [drm] Initialized i915 1.6.0 for 0000:00:02.0 on minor 0
[    8.016723] ACPI: video: Video Device [GFX0] (multi-head: yes  rom: no  post: no)
[    8.017127] input: Video Bus as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/LNXVIDEO:01/input/input4
[    8.254982] frecon[630]: Frecon using drm driver i915, version 1.6, date(0), desc(Intel Graphics) using atomic
[    8.559427] init: tpm2-simulator main process ended, respawning
[    9.245310] EXT4-fs (sda1): re-mounted 902cba86-a718-47f3-875a-4cdd9ffd5856 r/w. Quota mode: journalled.
[    9.357459] init: bluetoothd pre-start process (1185) terminated with status 1
[    9.470879] init: cr50-metrics main process (1400) terminated with status 1
[    9.477968] loop1: detected capacity change from 0 to 195528
[    9.484854] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    9.496679] Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    9.496878] Loaded X.509 cert 'wens: 61c038651aabdcf94bd0ac7ff06c7248db18c600'
[    9.582561] init: oobe_config_restore main process (1496) terminated with status 251
[   12.540826] init: ui main process (1393) terminated with status 2
[   12.556481] init: ui_stop_waiter main process (2184) killed by TERM signal
[   14.001601] init: oobe_config_restore post-start process (1497) terminated with status 143
[   14.843826] init: ui main process (2266) terminated with status 2
[   14.858372] init: ui_stop_waiter main process (2913) killed by TERM signal
[   17.054157] init: ui main process (2989) terminated with status 2
[   17.069790] init: ui_stop_waiter main process (3631) killed by TERM signal
[   19.254817] init: ui main process (3705) terminated with status 2
[   19.269915] init: ui_stop_waiter main process (4342) killed by TERM signal
[   23.545963] init: ui main process (4418) terminated with status 2
[   23.559361] init: ui_stop_waiter main process (5062) killed by TERM signal
[   25.305347] frecon[630]: Chrome did not release master. Trying again.
[   26.126034] frecon[630]: Chrome did not release master. Trying again.
[   26.907767] frecon[630]: Chrome did not release master. Trying again.
[   27.731488] frecon[630]: Chrome did not release master. Trying again.
[   28.569510] frecon[630]: Chrome did not release master. Frecon will steal master.
[   28.589647] init: ui main process (5139) terminated with status 2
[   28.605761] init: ui_stop_waiter main process (5785) killed by TERM signal
```