sebanc/brunch #1922: Audio doesn't work

Yunjonue, 2023-09-17:
Lenovo IdeaPad 3
Intel Core i5-1135G7 2.4GHz
8GB RAM
Booting from USB 3.2 32GB through Type-C port

Kernel 6.1, no Chromebook audio enabled, all drivers enabled in Chrome OS settings  menu

[image]
[image]

ZProLegend007, 2023-09-17:
Can you please post the output of `cras_test_client`?

Yunjonue, 2023-09-17:
System Volume (0-100): 100 
Capture Muted : Not muted
User muted: Not muted
Output Devices:
	ID	MaxCha	LastOpen	Name
Output Nodes:
	Stable Id	 ID	 Vol  UI       Plugged	L/R swapped	      Time Hotword	Type		MaxCha Name
Input Devices:
	ID	MaxCha	LastOpen	Name
	6	2	UNK		Post DSP Delayed Loopback
	5	2	UNK		Post DSP Loopback
	4	2	UNK		Post Mix Pre DSP Loopback
Input Nodes:
	Stable Id	 ID	Gain  UI       Plugged	L/R swapped	      Time Hotword	Type		MaxCha Name
	(ec289a19)	6:0	    0 1.000000     yes	            no	         0        	POST_DSP_DELAYED_LOOPBACK     X Post DSP Delayed Loopback
	(8c74f766)	5:0	    0 1.000000     yes	            no	         0        	POST_DSP_LOOPBACK     X Post DSP Loopback
	(8a04af91)	4:0	    0 1.000000     yes	            no	         0        	POST_MIX_LOOPBACK     X Post Mix Pre DSP Loopback
Attached clients:
	ID	pid	uid
	36	14295	1000
Num active streams: 0
Last audio active time: 1859, 727553236

ZProLegend007, 2023-09-18:
It seems that your output device isn't detected by cras (ChromeOS Audio Server). Did you say that you enabled chromebook audio? What recovery image are you using?

Yunjonue, 2023-09-18:
I tried with both of enabled and disabled chromebook_audio
I use Volteer, Voxel, 115 (which was suggested in the official documentation)

ZProLegend007, 2023-09-18:
Your device (I believe) originally runs Windows. Have you tried running linux on it before? I think there may be issues with drivers and/or soundcard detection. You need to figure out how to get your soundcard detected so that the system can output to it. Try to do some research online on similar issues for your device running linux online.

Yunjonue, 2023-09-18:
Yes, I did. My computer runs Windows, but I installed Linux. Thank you for advice

ZProLegend007, 2023-09-18:
Coolio, does sound work in Linux? As ChromeOS is built on Linux if the sound works on Linux alone then you could try install the same drivers on ChromeOS.

Good luck! Hope you figure things out.

Yunjonue, 2023-09-18:
Yes, it works on Linux. I'll try, thanks 🙂
Maybe also different recovery images can help? I can't figure out how to find a correct one for me, there's no difference, for example, between "rammus" and "volteer" at first sight, but they are for different CPU generations

ZProLegend007, 2023-09-19:
Try the "voxel" image. That image is said to work best on intel 11th gen devices.

leonardocustodiosilva, 2023-09-21:
My Samsung Galaxy Book2 with 12th Intel Core i5-1235U has no sound. Everything else works normally, including the microphone, only the sound doesn't work. I've already tried several images from Brya, now I'm testing with Primus, but it didn't work either.

[image]

ly9999999, 2023-09-22:
After downloading the sof firmware, refer to this method. https://github.com/sebanc/brunch/issues/1859#issuecomment-1689698613

agrimpelhuber, 2023-12-01:
And it might be worth checking out the additional steps here: https://github.com/sebanc/brunch/issues/1692

@ly9999999, hi again! I found that for my sound card, the sof firmware comes OK with current brunch (118) and kernel 6.1, but I need to run an extra script for  ALSA to fit.

Yunjonue, 2024-07-24:
with new Brunch version and kernel 6.6 audio appeared and works fine. don't need any additional drivers or parameters

closed 2024-07-24 by Yunjonue