sebanc/brunch #1925: entire brunch to grub menu

foodmaher, 2023-09-27:
Hello,

I followed the instructions for installation by using windows subsystem for windows.

Everything is done perfect except one thing.

I not prefer to download Grub2win to boot to brunch OS because i already have a ubuntu with grub which automatically installed during ubuntu installation. So, i want to add boot from partition D:/ChromeOS/chromeos.img to ubuntu grub but i don't know how i can edit the boot code as path of chrome OS image. I know my english not good but i hope you are able to understand what i mean :) 

menuentry "Brunch" --class "brunch" {
	img_path=/ChromeOS/chromeos.img
	img_uuid=7526c99d-f57e-4651-a1ba-b2caa9eccb78
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	source (loop,12)/efi/boot/settings.cfg
	if [ -z $verbose ] -o [ $verbose -eq 0 ]; then
		linux (loop,7)$kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params \
			cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path \
			console= vt.global_cursor_default=0 brunch_bootsplash=$brunch_bootsplash quiet
	else
		linux (loop,7)$kernel boot=local noresume noswap loglevel=7 options=$options chromeos_bootsplash=$chromeos_bootsplash $cmdline_params \
			cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path
	fi
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}

menuentry "Brunch settings" --class "brunch-settings" {
	img_path=/ChromeOS/chromeos.img
	img_uuid=7526c99d-f57e-4651-a1ba-b2caa9eccb78
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	source (loop,12)/efi/boot/settings.cfg
	linux (loop,7)/kernel boot=local noresume noswap loglevel=7 options= chromeos_bootsplash= edit_brunch_config=1 \
		cros_secure cros_debug img_uuid=$img_uuid img_path=$img_path
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}

So, how i can edit that to be able to booting from image without using grub2win just default ubuntu grub

ZProLegend007, 2023-09-28:
I gotchu my guy. I also used normal grub in my setup (As windows bootcamp can't access EFI on a Macbook.) Install grub-customiser and use that to add your custom script so you can boot. If you need help with anything the just comment :)

foodmaher, 2023-09-28:
> I gotchu my guy. I also used normal grub in my setup (As windows bootcamp can't access EFI on a Macbook.) Install grub-customiser and use that to add your custom script so you can boot. If you need help with anything the just comment :)

Now, i reinstall the brunch in Ubuntu rather than windows and everything works fine with default grub but when i shutdown the Chrome OS, my device not shutdown just the fan rotate faster and the device stick in black screen while the back light of keyboard still work. So, what is the problem and how i can force shutdown if that happen without using force shutdown by power button because that will damage my SSD and may losting my data. My CPU is Ryzen 3200u with 8 GB ram and 20 GB of img size.

ZProLegend007, 2023-09-29:
That also happened to me once, I fixed it by reinstalling brunch. I have force shutdown my device many many times with no adverse effects. Truth is it's not great for your ssd but its not terrible either, I think you will find a force shutdown doesn't affect your data. SSDs are very versatile. 

I recommend resinstalling and if that doesn't help then you may need to create another issue. I also recommend closing this issue as the inital problem is now fixed. Hope that helps!

closed 2023-10-02 by foodmaher