sebanc/brunch #1933: Tablet mode & Autorotation for Dell Latitude 5310 2-in-1

culjonito, 2023-09-30:
Hello,

First I will thank you that we can all use full ChromeOS on every device! Thats so cool!

I have a small issue with Tablet mode on my device "[Dell Latitude 53**1**0 2-in-1](https://www.dell.com/support/home/en-us/product-support/product/latitude-13-5310-2-in-1-laptop/overview)":
It's a convertible Device, so I can fold my screen 180 ° to use it as a tablet. The Keyboard itself is not dettachable.
On each Linux distributions and also on Windows 11 the operating system is switching automatically to Tablet mode when I fold the screen. But ChromeOS not. Only the keyboard is getting disabled, but no tablet mode enabled. I can manually enable the tablet mode, but the other problem is, that Autorotation is also not working. On other operating systems like Ubuntu it's working fine.
So my question to you is: Can you please fix this for us?

Just for your information: There is nearly the exact same Dell model with native Chrome OS pre-installed on the market - The "[Dell Latitude 53**0**0 2-in-1 Chromebook Enterprise](https://www.dell.com/support/home/en-us/product-support/product/latitude-13-5300-2-in-1-chrome-laptop/overview)" (codename: sarien)
For patching with Brunch i downloaded the sarien image, but I also tested it with volteer without success.

Side nodes:
Installed Brunch: **r116 20230928**
ChromeOS: **sarien R116**

Attached you can find "dmesg" and "lsmod" outputs.

I really hope you can help us :-)

Thanks in advance!

[lsmod.txt](https://github.com/sebanc/brunch/files/12775469/lsmod.txt)
[dmesg.txt](https://github.com/sebanc/brunch/files/12775470/dmesg.txt)

culjonito, 2023-10-02:
I just want to let you know, that it's working now.
The solution was simply to switch to the Chromebook 6.1 Kernel. Now everything seems to work fine.

DTayA1, 2023-12-01:
I have a (kinda) similar model (Latitude 3190 2-in-1), which also has a Chromebook twin. I used kernel 6.1, but I didn't get any sort of acknowledgement of the device's posture, as with your device on your original kernel. But what I find odd is that on 6.1 my camera+mic work fine, but on 5.10, only my mic and all of the device postures work. I have also had issues switching kernels and don't really want to install Brunch for the 5th or 6th time in the past few days. So how is the rest of your laptop?

closed 2025-12-18 by github-actions[bot]