sebanc/brunch #1934: Failure in shutdown ChromeOS

foodmaher, 2023-10-02:
I installed the brunch for the second time, and the problem still occurs when I shut down my laptop keyboard backlight, and the fan is still working while the laptop screen is still off.

I do not know why this problem happens to me while i have Ubuntu on my laptop, and i still get a similar issue with the sleep leds me to disabled this feature to avoid the infinite shutdown.

I will upload a video about my issue in the comments.

My laptop 

1) Ryzen 3200u
2) RAM 8GB 
3) SSD 1TB

My second inquiry, does my frequently force power off by a power button that will damage my SSD in terms of hardware or not, and if this behavior will damage my files from other partitions on my SSD ? Do you recommend me in using brunch, and if that harmful to my pc.

foodmaher, 2023-10-02:
[VIDEO FOR THE PROBLEM ](https://youtu.be/qG9vPu9BoTg)

ilovefreesw, 2023-10-03:
turn off usb power and maybe fast boot in BIOS settings.

foodmaher, 2023-10-03:
> turn off usb power and maybe fast boot in BIOS settings.

But if that will affect on windows or not

foodmaher, 2023-10-03:
> turn off usb power and maybe fast boot in BIOS settings.

I disabled fast boot, but I can't know what is power by usb (I think is presence in alternative name but it is disabled), but i think it haven't relating to my issue because the problem happened when i shutdown my laptop. 

last i found resume pc work by lid which depends on S3 state. So, is that also have relating to my problem or not?

foodmaher, 2023-10-04:
The problem is solved by turning off fast boot; even Windows boot time is shorter without the fast boot option, and I tested suspend in Ubuntu, which works without any catch. So, I will keep this issue open for an extra three days. So, if anything is right now, I will close it.

closed 2023-10-08 by foodmaher