sebanc/brunch #1935: ChromeOS Stuck logo

mehedi6254, 2023-10-04:
I have DELL Inspiron 3576 
-> Intel core i5 8th gen
-> AMD Radeon Graphics Dedicated 2GB
I can successfully boot till version 107. But after that version the newer version stuck on chrome os white logo dark screen after boot up. Nothing happens after plenty of time.
Is there any way to fix this issue on newer build like 115, 116 ?
Thanks a ton

gittelitt, 2023-10-08:
I also have a 8th gen Intel CPU and a AMD GPU. I'm experiencing the same issue

As2Bax, 2023-10-12:
are you using the right build for the processor?

neptalactone, 2023-11-01:
Hi Guys,

As for this issue, 

First make sure you're using the right recovery image for your Processor Version. i assume it is caused by your dedicated GPU that different vendor from your CPU so ChromeOS detected both your integrated GPU from intel and dedicated GPU from AMD. This will make ChromeOS "Confused" and pick, usually dedicated GPU, but on some Laptop, the display monitor usually connected to Integrated GPU and render it using dedicated GPU. 

The potential workaround to fix this issue is to disable your dedicated GPU. if your laptop support to disable dedicated GPU from BIOS, you can disable it. But if there is no option or feature in the BIOS to disable it, you can use this experimental kernel parameter that you can enter from brunch settings in grub menu: 
- remove_dgpu=0000:XX:XX.X

As for the step to use it:
- try to enter tty from the stuck logo with combination key ctrl+alt+f2
- log in as chronos user, then
- Identify your dGPU pci address in "sudo lspci" (first column, in the form XX:XX.X).
- Add "remove_dgpu=0000:XX:XX.X" as an additional kernel command line parameter. (replace XX:XX.X with the pci address of your dedicated gpu)

This parameter is referred to by sebanc on my issue post in #1904 . You can try to use it.