sebanc/brunch #1936: displaylink driver fails to start in R116 stable

weberk, 2023-10-06:
Just updated from R113 to R116  and realized the following in dmesg:
```
[12330.983240] usb 2-1.1: Product: ThinkPad USB 3.0 Ultra Dock                                                          
[12330.983241] usb 2-1.1: Manufacturer: DisplayLink                                                                     
[12330.983243] usb 2-1.1: SerialNumber: 4551152301011272M2VK                                                            
[12331.040180] usbcore: registered new interface driver cdc_ether                                                       
[12331.068259] init: displaylink-driver pre-start process (5385) terminated with status 1 
```

ZProLegend007, 2023-10-07:
Try the latest brunch-unstable build. It may contain a fix. https://github.com/sebanc/brunch-unstable/releases/tag/r117-unstable-20231007

weberk, 2023-10-07:
Fantastic! It is working again in [r117-unstable-20231007]! Thanks for pointing out or fixing.

weberk, 2023-10-07:
Fixed in [r117-unstable-20231007](https://github.com/sebanc/brunch-unstable/releases/tag/r117-unstable-20231007)

closed 2023-10-07 by weberk