sebanc/brunch #1937: Apple drivers missing in R116 stable

weberk, 2023-10-06:
Since updating from R113 to R116 my fan runs on max all the time, keyboard backlite not working, keyboard layout not in line with printed caps.

It seems some apple drivers are not running like hid_apple?

Any hints on how to workarround this effects?

ZProLegend007, 2023-10-07:
Hey @weberk,  could you tell me what model Mac you are using?

weberk, 2023-10-07:
I use a 
Hardware name: Apple Inc. MacBookAir6,1/Mac-35C1E88140C3E6CF, BIOS 118.0.0.0.0 08/06/2019
It is  the mid 2017 model.

ZProLegend007, 2023-10-08:
Coolio, are you sure the Hardware name is right? The MacBookAir6,1 is a 2014 model. What kernel are you using?

weberk, 2023-10-08:
Well I got that information from dmesg output. But even Apple considers mid 2017 MacBookAir 11 inch as MacBookAir6,1 : https://support.apple.com/de-de/HT201862 

My kernel is: 
```
crosh> shell
chronos@brunch / $ uname -a
Linux localhost 6.1.55-brunch-sebanc #1 SMP PREEMPT_DYNAMIC Sat Oct  7 14:15:42 UTC 2023 x86_64 Intel(R) Core(TM) i7-4650U CPU @ 1.70GHz GenuineIntel GNU/Linux
chronos@brunch / $dmesg|grep MacBookAir
[    0.000000] DMI: Apple Inc. MacBookAir6,1/Mac-35C1E88140C3E6CF, BIOS 118.0.0.0.0 08/06/2019
chronos@brunch / $
```

ZProLegend007, 2023-10-08:
Hmm, on that website it shows these,

MacBook Air (13-inch, 2017)
Model Identifier: MacBookAir7,2

MacBook Air (11-inch, Mid 2013)
Model Identifier: MacBookAir6,1

MacBook Air (11-inch, Early 2014)
Model Identifier: MacBookAir6,1

Anyways besides the point. Sorry I should have rephrased my previous question. What brunch kernel are you using? (e.g t2-macbook, 6.1, chromebook 6.1)

weberk, 2023-10-08:
Brunch kernel is listen above from the response of the comman **uname -a** :
```
chronos@brunch / $ uname -a
Linux localhost 6.1.55-brunch-sebanc #1 SMP PREEMPT_DYNAMIC Sat Oct  7 14:15:42 UTC 2023 x86_64 Intel(R) Core(TM) i7-4650U CPU @ 1.70GHz GenuineIntel GNU/Linux

```
As you can sse it is the stable Brunch r117 20231007 from yesterday.

weberk, 2023-10-08:
So it remains to check if hid_apple can be added back. In the R117 distrubution its not in:
```
brunch /lib/modules/6.1.55-brunch-sebanc # find . -name '*apple*'
./build/arch/arm64/boot/dts/apple
./build/arch/arm64/include/asm/apple_m1_pmu.h
./build/drivers/soc/apple
./build/drivers/net/ethernet/apple
./build/drivers/net/appletalk
./build/include/linux/platform_data/x86/apple.h
./build/include/linux/apple-gmux.h
./build/include/linux/soc/apple
./build/include/linux/apple_bl.h
./build/include/linux/apple-mailbox.h
./build/include/dt-bindings/interrupt-controller/apple-aic.h
./build/include/dt-bindings/pinctrl/apple.h
./build/sound/soc/apple
./build/net/appletalk
./kernel/drivers/input/mouse/appletouch.ko
./kernel/drivers/input/keyboard/applespi.ko
./kernel/drivers/usb/misc/appledisplay.ko
./kernel/drivers/hwmon/applesmc.ko
./kernel/drivers/video/backlight/apple_bl.ko
./kernel/drivers/hid/hid-appleir.ko
./kernel/drivers/platform/x86/apple-gmux.ko
```

ZProLegend007, 2023-10-09:
Nice changing the model on every comment 😂. You may just need to wait for @sebanc to add those back in.

weberk, 2023-10-09:
Sorry for my updates. Did not want to offend you, but to correct the information needed for solving the issue. 

Thank you for pointing out that my kernel output was saying 2013 hardware, while I was claiming 2017. Time flies...

ZProLegend007, 2023-10-09:
All good mate, have you tried changing the software keyboard layout?

weberk, 2023-10-09:
Keyboard Layout is already german. There is no special Mac german layout available. It seems its only one key different between the Mac Layout and the PC key layout:
`<,>,|    vs.    ^°`

ZProLegend007, 2023-10-09:
Hmm, I think it's just something else to wait for sebanc to re-add. There are a lot of MacBook drivers that seem to have gone missing.