sebanc/brunch #1939: How to remove Brunch?

Naazim-Apps, 2023-10-09:
Hi, I loved the Brunch project but I want to go back to Windows. I would like to remove Brunch but can't get to BIOS. I can however access the GRUB command line. Can I get help here? @sebanc I saw you helped someone else but it's not working. Pictures below.
[image]

[image]

ZProLegend007, 2023-10-09:
How did you install brunch? What device are you using and what OSs are installed, then we can go from there :) Did you set it up so that you can only boot brunch?

Naazim-Apps, 2023-10-10:
> How did you install brunch? What device are you using and what OSs are installed, then we can go from there :) Did you set it up so that you can only boot brunch?

If you Google "Brunch beebom tutorial" and click the Beebom link from 2022, those are the steps I followed.

I am using a HP laptop with 4GB RAM, and an Intel Celeron N4020 processor.
Windows 11 was installed before, but I completely replaced it with Brunch.

And yeah, I set it up so I can only boot brunch.

ZProLegend007, 2023-10-12:
You need to make a windows installer USB and then boot into that to install Windows 11 again. Also you may need to access BIOS to do that so look around online for the specific keys that need to be pressed on your device to access that.

Naazim-Apps, 2023-10-12:
> You need to make a Windows installer USB and then boot into that to install Windows 11 again. Also you may need to access your BIOS to do that so look around online for the device specific keys that need to be pressed during boot to access that. (It's usually a function key so try mashing them a few times while booting.)

Yes, I have tried to get into BIOS. I know the key, I've used it before, but it's like the Brunch boot screen is as far as I can get to BIOS.

Naazim-Apps, 2023-11-01:
Hey, just wanted to let you know that I finally got to BIOS, thanks for trying to help.

ZProLegend007, 2023-11-02:
Good to hear! Make sure to close the issue if the problem is solved :)

closed 2023-11-04 by Naazim-Apps