sebanc/brunch #1943: Android subsystem not working with Chrome OS 117 and brunch r117

As2Bax, 2023-10-12:
Play store will not start on a new install it just pops up with "Something went wrong"
I have tried both android_init fixes
I am using the "grunt" recovery image

Specs:
Model: Lenovo Ideapad 130-15AST
CPU: AMD A6-9225
RAM: 4GB
SSD: PNY CS900 500GB

ZProLegend007, 2023-10-12:
Try using a different recovery image, sometimes brunch may encounter issues with other images. I recommend rammus (shyvana) cros.tech/device/shyvana/

As2Bax, 2023-10-12:
> Try using a different recovery image, sometimes brunch may encounter issues with other images. I recommend rammus (shyvana) [linked here](https://cros.tech/device/shyvana/)

hmm but that's for a Intel Chromebook not an AMD one won't that cause issues?

ZProLegend007, 2023-10-12:
Ah sorry, I glossed over the details. Try this zork image (gumboz) [linked here](https://cros.tech/device/gumboz/)

As2Bax, 2023-10-12:
The issue is fixed, I tried that version, and the Play Store works.

closed 2023-10-12 by As2Bax

Mo-5590, 2025-01-13:
For me i use chromeOS Flex and no PLAY STORE appear or work on it
Until i install chromeOS Rammus 131 with brunchr131
and it works 100/100

Great thanks to the team of Brunch here

[image]