sebanc/brunch #1945: chrome os 117 don't boot

naiimnasr11, 2023-10-18:
hello everybody
i can't install chrome os 117 chromeos_15572.57.0_rammus_recovery_stable-channel_mp-v3
on me pc i7 4790k hp ProDesk 600 g1 blocked in logo chrome os
thanks for help.

stanley211, 2023-10-19:
You probably not using a compatible chrome os recovery  version or for your cpu type. Like Zork type is for AMD, but also have to closes match the the Computer brand, Like HP zork.

naiimnasr11, 2023-10-19:
thanks stanley211
I use board: rammus shyvana 
then what is the compatible for me pc i7 4790k hp ProDesk 600 g1

stanley211, 2023-10-19:
Try Chromebook x360 14 version recovery 

And try to read this brunch post tutorial at:
https://github.com/sebanc/brunch/discussions/982
"Always use at your own Risk"

naiimnasr11, 2023-10-20:
thanks stanley211
i am using Nami image its on

closed 2025-12-16 by github-actions[bot]