sebanc/brunch #1946: how i can use virtual terminal 2  in chrome os 117

naiimnasr11, 2023-10-20:
hello everybody
how i can use virtual terminal 2  in chrome os 117
i have this message after pressing CTRL+ALT+T
crosh> shell
Sudo commands will not succeed by default.
If you want to use sudo commands, use the VT-2 shell
(Ctrl-Alt-{F2/Right arrow/Refresh}) or  the image with the
login_enable_crosh_sudo USE flag:

$ USE=login_enable_crosh_sudo emerge-$BOARD chromeos-login
or
$ USE=login_enable_crosh_sudo cros build-packages --board=$BOARD

bybenjamin, 2023-10-21:
I can root with Ctrl-Alt-F2 keys, but I cannot use brioche. How to fix this problem ?

markr1971, 2023-11-08:
how do you copy and paste in Terminal 2, so you can update Brunch & ChromeOS 

To manually update Brunch and ChromeOS together:

Download the [latest Brunch release](https://github.com/sebanc/brunch/releases/latest)
Download the [latest recovery](https://cros.tech/) matching your install and extract the bin.
Open a TTY with Crtl + Alt + F2 and login as chronos.
Run the built in command to update Brunch.
Replace brunch_archive.tar.gz with the file's actual filename.
Replace recovery.bin with the file's actual filename.
sudo chromeos-update -r ~/Downloads/recovery.bin -f ~/Downloads/brunch_archive.tar.gz

Restart ChromeOS after the update finishes.

Thanks