sebanc/brunch #1949: Chuwi HI8 is  stuck on ChromeOS logo

ldrago63, 2023-10-24:
Hello and first thank you for your work,

I'm trying to give a second life to a chuwi hi 8 tablet by installing Chrome OS.

SoC: Intel Atom Z3736F
CPU: Intel Bay Trail, 2160 MHz
Cores: 4
GPU: Intel Gen 7, 646 MHz
RAM: 2 GB

First I tested with ChromeOS Flex on a bootable usb key. Touchscreen wokrs but no sound, no bluetooth.

After so I turned to brunch and install ChromeOS using shyvana (117). My tablet is stuck on ChromeOS logo after installation. I tried with several kernel versions.

Could someone please tell me the best recovery to use with the best Kernel?

Thank you and have a good day :)

bybenjamin, 2023-10-25:
you should try with snappy recovery

ldrago63, 2023-10-25:
Thank you @bybenjamin.

Good news, I have tested this morning using shyvana r107 and brunch r107. Without any option, touch, sound work.

I am searching for auto auto brightness and bluetooth in brunch option. If all work, I will add details on device page.

ldrago63, 2023-10-25:
After multiple tests:

r107 and brunch r107:
- no bluetooth (RTL8723BS) : can be enable but can't find devices
- no auto rotation

ldrago63, 2023-10-30:
Hi,

Able to boot using r107.

closed 2023-10-30 by ldrago63