sebanc/brunch #1954: Surface Pro 3 - Updating to Rammus 117 results in stuck at ChromeOS logo

venturis62, 2023-10-30:
I've been successfully running Brunch Framework / ChomeOS (Rammus) on my Surface Pro 3 for some time.

Recently tried updating to 117 but after updating Brunch to the latest version (117).  However, after restarting, the boot process gets stuck at the ChomeOS logo.

I've tried both upgrading from 116 -> 117 and also a clean installation directly to 117.  Both give the same results.  No boot and stuck on ChomeOS logo.

Tried a number of different startup options including different kernels.  On a couple of occasions have hit the ChomeOS system "repair" which I've let finish but no change after the restart.

Tried starting up with the debug verbose on however, the boot sequence finishes completely and the screen changes to the ChomeOS logo before hanging.

Any help appreciated.

XenHat, 2023-11-05:
My surface pro 3 won't boot anything newer than 114.

presslab-us, 2023-11-11:
I have the same problem on my two devices (Lenovo Yoga 2 Pro, Dell Venue 8 Pro).  v116 is working, v117 (and v118) is stuck at the ChromeOS logo.  This is using v118 Brunch for all.

I suspect it's a display issue because pressing the power button I can see disk activity (thumb drive LED) so it's not frozen.  Both of these devices are pretty old (~2014), which is similar for the Surface Pro 3.

I tried other kernels with no success.

`verbose` works to get more debug info before ChromeOS starts; is there a way to get a debug output after it starts?

venturis62, 2023-11-14:
Tried using the "Octopus" v118 recovery image with the latest brunch framework (118) as it has been reported to work better on some older hardware however, got exactly the same result.

At this stage it appears v116 recovery is as far as this old lady will go.

pkbw2000, 2024-01-26:
ASUS Transformer Mini R107HAF
Intel® Quad-Core Atom™ x5-Z8350 Processer
4gig RAM
Using Rammus

I also encountered the same problem, i'v tried R117-R120 .
and R116 all fine but playstore not working.WHY??

closed 2026-01-18 by github-actions[bot]