sebanc/brunch #1958: Touchscreen Issues on Surface pro 8

thelife2face, 2023-11-01:
Hi everybody.
I'm install Brunch and Chrome OS (version 117) on my tablet (Surface pro 8).
Most functions work well but touchscreen is not working.
I has tried option: iphc_touchscreen serveral times, on kernel 5.15 and 6.1. Sometime, touchscreen is working, but when i reboot system, touchscreen is not woking again.
If anyone has any ideas to help me. Thank
(Sorry, my English isn't good)

Datadelete10426, 2023-11-04:
Enter the directory `/lib/module/#uname -r#/`, replace `#uname -r#` with your own kernel version, for instance `6.1.60-brunch-sebanc`.
`cd /lib/module/6.1.60-brunch-sebanc`
Unload the ithc module.
`sudo rmmod ithc`
Reload the ithc module.
`sudo insmod ithc.ko`
If it still doesn't work, try several more times until the touchscreen function is restored.

thelife2face, 2023-11-18:
> Enter the directory `/lib/modules/#uname -r#/`, replace `#uname -r#` with your own kernel version, for instance `6.1.60-brunch-sebanc`. `cd /lib/modules/6.1.60-brunch-sebanc` Unload the ithc module. `sudo rmmod ithc` Reload the ithc module. `sudo insmod ithc.ko` If it still doesn't work, try several more times until the touchscreen function is restored.

Can you give specific instructions?