sebanc/brunch #1959: stuck at ChromeOS logo, black screen

YazWare, 2023-11-02:
I have an HP Stream 14-ax016nf
- Intel Celeron N3060 (Braswell)
- Intel HD Graphics 400
I used the octopus recovery r118, brunch r118.
Everything boots properly until I get into the ChromeOS logo, it gets stuck for hours. I'm booting from an USB.
It's my first time with Brunch and ChromeOS so maybe that's something obvious but I don't really know a lot about these.

XenHat, 2023-11-03:
Duplicate of many, many issues here.

Try booting ChromeOS 114 on a cpu this old. my two machines with Haswell processors refuse to boot or update to anything newer.

YazWare, 2023-11-03:
thank you, working fine now

closed 2023-11-03 by YazWare

Haxker-00, 2024-09-05:
for intel i7 2nd gen??

Haxker-00, 2024-09-05:
> Duplicate of many, many issues here.
> 
> Try booting ChromeOS 114 on a cpu this old. my two machines with Haswell processors refuse to boot or update to anything newer.

for 97 2nd gen??