sebanc/brunch #1963: RTL8821CE Not Working Correctly

nokidevz, 2023-11-05:
Hello! I have a Laptop and i am dualbooting Ubuntu 22.04 and ChromeOS 118. So when i boot into ChromeOS, i connect to my Wi-Fi, but i realised that only 2 WiFi networks were available, even though theres 20+ on Windows or or my phone, and the signal on my network was very low, but when i connected to it, i noticed it was randomly disconecting me from my WiFi, randomly the speed would go in very low speeds and so on, i did also have this exact same issue on my Ubuntu installation, but i just install Tomaspinho's rtl8821ce driver and blacklisted rtw88_8821ce, rtw88_8821c, rtw88_core, rtw88_pci and mac80211, and now on my Ubuntu installation, everything's working just fine, i have strong signal, no disconnecting, and all WiFi networks i have on my phone or had on my Windows are here. Now i did do some research, and i did enable the "rtl8821ce" option from Brunch settings and added the options=rtl8821ce it doesnt do anything. Thanks in advance.

nokidevz, 2023-11-06:
Hello! I fixed this issue, if anyone else is experiencing this same issue, what i did was:

1. Enable rtl8821ce in Brunch (settings)
2. Installed Chromebrew
3. Using Chromebrew i installed git, make, gcc, bc, binutils
4. I cloned the RTL8821CE Driver from Tomaspinho using git
5. Then cd into the cloned folder
6. and ran this using CTRL+ALT+T ln -s /usr/local/lib64/libreadline.so.8 /usr/local/lib64/libreadline.so.6 
because ChromeOS seems to use libreadline.so.8 instead of libreadline.so.6
7. After that you need to type make all
8. Then get into the VT-2 Terminal by doing CTRL+ALT+F2
9. cd *directory of rtl8821ce driver*
10. sudo insmod 8821ce.ko
11. sudo rmmod rtw88_8821ce
12. sudo rmmod rtw88_8821c
13. sudo nano /etc/modprobe.d/rtw88_blacklist.conf
14. And in here type:
15. blacklist rtw88_8821ce
16. blacklist rtw88_8821c
17. Then CTRL+X, Yes and Enter
18. After that in Crosh Terminal check if the driver is running by doing lsmod | grep 8821ce
19. The last line should be like this: cfg80211              757760  4 wl,rtw88_core,mac80211,8821ce
20. If you followed all steps, your driver SHOULD work, for me it did.

billabongbruno, 2024-02-11:
> Hello! I fixed this issue, if anyone else is experiencing this same issue, what i did was:
> 
>     1. Enable rtl8821ce in Brunch (settings)
> 
>     2. Install Chromebrew
> 
>     3. Using Chromebrew i installed git, make, gcc, bc, binutils
> 
>     4. I cloned the RTL8821CE Driver from Tomaspinho using git
> 
>     5. Then cd into the cloned folder
> 
>     6. and ran this using CTRL+ALT+T ln -s /usr/local/lib64/libreadline.so.8 /usr/local/lib64/libreadline.so.6
>        because ChromeOS seems to use libreadline.so.8 instead of libreadline.so.6
> 
>     7. In the terminal run: make all
> 
>     8. Then get into the VT-2 Terminal by doing CTRL+ALT+F2
> 
>     9. cd _directory of rtl8821ce driver_
> 
>     10. sudo insmod 8821ce.ko
> 
>     11. sudo rmmod rtw88_8821ce
> 
>     12. sudo rmmod rtw88_8821c
> 
>     13. sudo nano /etc/modprobe.d/rtw88_blacklist.conf
> 
>     14. And in here type:
> 
>     15. blacklist rtw88_8821ce
> 
>     16. blacklist rtw88_8821c
> 
>     17. Then CTRL+X, Yes and hit enter
> 
>     18. After that in Crosh Terminal check if the driver is running by doing lsmod | grep 8821ce
> 
>     19. The last line should be like this: cfg80211              757760  4 wl,rtw88_core,mac80211,8821ce
> 
>     20. If you followed all steps, your driver SHOULD work, for me it did.

Hello.

Sorry to bother you, but would you, by any chance, be willing to help me do the same procedure for my Wifi card (or one of my USB adapters)?

I have a Surface Pro 4 i5 running brunch-unstable 121 with ChromeOS shyvanna 121 and I'm facing the same wifi issues.

I have the Marvell Avastar 88W8897 integrated WIFI/BT board and I also have three WIFI USB adapters (because of another OS), which are:

1 - COMFAST CF-811AC (RTL8811)
2 - TP-Link Archer T2U Nano (rtl8811au / rtl8811cu)
3 - TP-Link TL-WN725N (RTL 8188)

Is there anyway you could help me get one of these devices working (preferrebly the integrated Marvell Avastar, but if not possible, any of the others)?

Sorry to bother you and thank you in advance.

Best regards,
Bruno.