sebanc/brunch #1966: Please make a module for m1 macbookair!

kawadaka, 2023-11-12:
Spit out module error. Please make a module!
The model is a macbookair 2020 m1 chip. i was able to install it from terminal after installing with ubuntu asahi.

kawadaka, 2023-11-12:
[image]

kawadaka, 2023-11-12:
[image]

kawadaka, 2023-11-12:
[image]

ZProLegend007, 2023-11-14:
It's not that simple. M1+ is using an entire different architecture. Brunch is built for 64_86x PCs. This won't be possible unless an ARM revision of brunch is made with an ARM compatible kernel.

wannafedor4, 2024-07-01:
We should start with #1756. This would be relatively easy with @AppleWOA's UEFI implementation, once that's done. The Apple Windows-On-ARM thing requires UEFI to boot Windows, so I we use that all that's left is an ARM version of the project, and some images of chromeOS for high-end ARM Chromebooks. Anything else could be done with the open source chromium utilities, or generic open source tools. And any small thing that's between boot loader and chromeOS would be fixed with Apple's _extremely_ fast Rosetta 2.