sebanc/brunch #1968: Wifi broken on ChromeOS 119

billabongbruno, 2023-11-20:
Hey there.

So, first of all, I would like to thank you for all of your hard work that you have put into this project in the pursuit of making ChromeOS run on non-native machines.

Thank you.

As for the issue itself, I am aware that it's probably a ChromeOS issue itself, but this is just to ask for help and not criticize your work.

One of my machines running ChromeOS via Brunch is a Surface Pro 4, which has the infamous Marvell Avastar 88W8897 Wifi / BT board.
Up to and including both the STABLE and UNSTABLE Brunche versions 118 (Brunch r118 stable 20231101 & Brunch r118 unstable 20231112) everything worked perfectly, including Wifi, while running up to and including ChromeOS version 118.

However, everytime I tried to update to ChromeOS 119 via Settings, Wifi no longer worked, no matter the Kernel I chose.
And yes, I tried them all.
I am using the shyvanna rammus_recovery (I have an Intel processor).

The board itself seems to be recognized, as I do have an On/Off toggle in both the flyout menu and Settings itself.
It just doesn't find any network and won't even connect to anything even if one is manually added.
It's just stuck in a loop where it's trying to find networks, but fails to do so everytime.

Interestingly enough, I also own a Archer T2U Nano Wifi dongle (because of Hackintoshing) and even that adapter won't connect to anything, even after I enabled its driver in Brunch config (RLT8812AU).

Is there something that can be changed, tweaked or adjusted to make it work, or is ChromeOS 118 the last release that I'll be able to use?

Best regards,
Bruno.

JiayuanWen, 2023-11-22:
Same issue here with an Acer Chromebook 15 (CB315-1HT) with Brunch r118 stable 20231101. Also tried different kernels + config combos with no dice. Downgrading to 118 for the time being.

rama982, 2023-11-26:
same, something with crypto ig

Arefu, 2024-04-29:
I see same issue, but I can see my wireless network, though I am unable to connect to it. 
Claims incorrect password, despite trying a hotspot as well

S-Jay44, 2024-06-24:
Broken WIFI and bluetooth Chrome OS brunch 125 - Lenovo Ideapad 530S

At startup, I newly selected "ChromeOS (settings)" and the process of changing the kernel and classic fixes starts. I have kernels 6.6, 6.1 and 5.15 for selection and chromebook - 6.6 etc. I selected all classic kernels and selected from the options "broadcom_wl" and then just enter, enter...restart..repair and after starting again and wifi still does not work. It doesn't detect any nearby devices, and if it sees my wifi for a while, it fails after a while. The same bluetooth. They also can't see anything around them. Please, does anyone have any ideas... I like the way chrome OS looks and works, but without wifi the laptop is unusable 😦

Sciddie, 2025-01-22:
> I see same issue, but I can see my wireless network, though I am unable to connect to it. Claims incorrect password, despite trying a hotspot as well

Same thing here... in my case it works to connect to a WiFi without password protection.
Did anybody find a solution?

closed 2025-12-14 by github-actions[bot]