sebanc/brunch #1977: Stuck at "Chrome OS" screen

leighroy, 2023-12-07:
Lenovo Thinkcentre 
Intel Core i5-4590T
16gig RAM

Anything after 116 the computer just sticks. Could it be just a thing of the CPU getting old? 
Checked other forums on here and dgpu thing doesn't apply here and there doesn't seem to be other fixes. 
Any advice would be much appreciated. 
Cheers

tejasraman, 2024-01-10:
> Lenovo Thinkcentre Intel Core i5-4590T 16gig RAM Using Rammus
> 
> Anything after 116 the computer just sticks. Could it be just a thing of the CPU getting old? Checked other forums on here and dgpu thing doesn't apply here and there doesn't seem to be other fixes. Any advice would be much appreciated. Cheers

What kernels and recoveries have you tried? Sometimes all it takes to get Brunch to work is finding the right recovery image and kernel version :D

pkbw2000, 2024-01-26:
ASUS Transformer Mini R107HAF
Intel® Quad-Core Atom™ x5-Z8350 Processer
4gig RAM
Using Rammus

I also encountered the same problem, i'v tried R117-R120 .
and R116 all fine but playstore not working.WHY??

Nikto4242, 2024-02-10:
> ASUS Transformer Mini R107HAF Intel® Quad-Core Atom™ x5-Z8350 Processer 4gig RAM Using Rammus
> 
> I also encountered the same problem, i'v tried R117-R120 . and R116 all fine but playstore not working.WHY??

Same in Asus X451 Pentium N3170

closed 2025-12-14 by github-actions[bot]