sebanc/brunch #1978: Troubles with Tablet Mode

phitschhd, 2023-12-10:
Hi,
first of all thank you for this great project!
Device: HP Elite X2 G8 (with detachable keyboard)
Describe Problem: When i detach my keyboard, then ChromeOS goes correctly in the tablet-mode, but the onscreen keyboard doesn’t appear and the auto rotate is also not available. If i activate the force_enable_tablet mode in the brunch framework settings the tablet mode works perfectly (with onscreen keyboard and also the rotation) but the detachable keyboard doesn’t work when i connect it to the tablet (except the f keys they work)

phitschhd, 2023-12-16:
Update: I managed it to work properly by adding the right kernel (04F3:0C79) for my device in the /usr/udev/rules.d/99-tablet-mode.rules file.
@sebanc could you please add 
ACTION=="add", SUBSYSTEMS=="hid", KERNEL=="*:04F3:0C79.*" , RUN+="/bin/bash -c 'echo 0 > /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode'" 
ACTION=="remove", SUBSYSTEMS=="hid", KERNEL=="*:04F3:0C79.*" , RUN+="/bin/bash -c 'echo 1 > /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode'" in the 99-tablet-mode.rules file so that I don't have to change the kernel every time I do a brunch update?

sebanc, 2024-01-01:
Hi,

Your patch is included in brunch r119 stable release.

phitschhd, 2024-01-01:
Hi, thanks, appreciate your work! :)

closed 2024-01-01 by phitschhd

tenda96, 2024-06-27:
hi could you please add mine too?

ACTION=="add", SUBSYSTEMS=="hid", KERNEL=="*:044E:1218.*" , RUN+="/bin/bash -c 'echo 0 > /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode'"
ACTION=="remove", SUBSYSTEMS=="hid", KERNEL=="*:044E:1218.*" , RUN+="/bin/bash -c 'echo 1 > /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode'"

Thank you and thanks for your work!
Let me know :)