sebanc/brunch #1986: Surface Pro 5 - Auto rotate won't work

datascientyst, 2023-12-28:
Using Surface Pro 5 2017 with Brunch. First taste of Chrome OS and thanks for this lovely work.
Two issues, one of which is a  problem - Auto-rotate does not work even after disconnecting all keyboards / mice (wireless too). The tablet mode shows up but when I rotate the device it stays in landscape mode.

kolle86, 2023-12-28:
You have to enable **force_tablet_mode** in the framework options.

datascientyst, 2023-12-29:
Thanks. However, I don't have issue with the tablet mode, it works flawlessly when I remove the keyboard and mouse. Would force_tablet_mode solve the auto-rotate issue while not having any negative impact?

kolle86, 2023-12-29:
Yes. With that Option enabled you will have auto rotation working. I did not notice any negative impact.

closed 2023-12-29 by datascientyst